odriverobotics/ODrive #407: index (Z) signal (index_pin) only pulldown

AKEOPLUS-boris-bocquet, 2020-05-10:
Dear madcowdswe, We are currently using your odrive board and your codes   (firmware w-v0.4.12). We lost a lot of time when searching for the index (Z) signal. Indeed we linked the Z of our encoder, which is pullup (level high when the encoder is on the index). The motor was not spinning when calling ```AXIS_STATE_ENCODER_INDEX_SEARCH```, and for good reasons : signal was low. We understood that because the motor was not spinning at all, or just during acceleration phase, when playing with the parameter ```find_idx_on_lockin_only```. Then, after reading at ```Encoder::set_idx_subscribe```, we saw that you suscribe the GPIO to a pulldown level. So we linked /Z instead of Z and now we think we have the expected behaviour.
Can you please : 
* Confirm our analysis and conclusions
* Improve the documentation (e.g. on encoders.md)
* If possible, have a parameter in a config that can change the default ```GPIO_PULLDOWN``` value

Thanks in advance, the AKEOPLUS team.

AKEOPLUS-boris-bocquet, 2020-05-12:
Sorry : it was actually written [in your getting started file](https://github.com/madcowswe/ODrive/blob/master/docs/getting-started.md#wiring-up-the-encoders). Maybe this is good to have it in your encoder.md file too. And be able to change it in the config.

samuelsadok, 2020-05-19:
The `GPIO_PULLDOWN` that you're seeing actually just enables the internal pull-down resistor of the microcontroller but since the on-board resistor is stronger (3.3kΩ vs 40kΩ) this should have no effect.
In `GPIO_subscribe` you can see `GPIO_InitStruct.Mode = GPIO_MODE_IT_RISING;`, so the index pin reacts to a rising edge.
What encoder are you using? As far as I can see the only encoder type that would not work with is if the index output is high-impedance in inactive state and high (at low impedance) in active state. Is that what your encoder is like?

AKEOPLUS-boris-bocquet, 2020-07-08:
Hi. Sorry for the very late reply. We used a custom made eletronic and it provides a signal from an output (with a "transistor"). So yes, this electronic is not like the usual encoders. On a next revision, we will do just like what you propose. Sorry for the inconvenience. Please considere if possible my improvement proposal. Best Regards.

closed 2020-07-08 by AKEOPLUS-boris-bocquet