odriverobotics/ODrive #409: No check for CAN extended frame ID vs. non extended frame ID

r91andersson, 2020-05-13:
In our project we have two different sets of motor controllers, one pair of Odrives and one pair of VESC's.
All of them are connected to the same CAN-bus. VESC are using extended CAN ID, and ODrive are using non-extended frame ID. 

We noticed when we run our motor controllers together that for one motor connected to one of the the Odrive's, the E-Stop was triggered. The CAN id was 0 for this motor.
We found out that if an extended CAN frame is sent on the bus, it will be read by the Odrive, and its id will be recognized as 0 (thus accepting the command that was intended to be sent to the VESC controller). 

So to correct for this, use the the flag "isExt", and do only allow incoming messages once this flag is false. We made a quick fix for this by selecting a node ID != 0 for the motor.

samuelsadok, 2020-05-18:
Thanks for the report. Extended CAN IDs were indeed just clipped of their upper 18 bits and then treated like standard CAN IDs. I figured I'd just add proper support for extended CAN IDs, see #411. If you find the time you can check if it works in your setup.

kalvdans, 2020-05-18:
long and short CAN addresses can exist on the same bus, since there is a bit that differentiates them in the protocol. There should in theory be no need to support extended CAN IDs in the ODrive code.

https://en.wikipedia.org/wiki/CAN_bus#Base_frame_format

samuelsadok, 2020-05-19:
Yeah I'm aware, I just figured if it's quick to add, why not.

closed 2020-05-20 by samuelsadok