odriverobotics/ODrive #416: Add thermistor support

rowanG077, 2020-05-28:
I would like to have support for reading thermal data of a ntc/ptc and setting a thermal limit. I want to implement this myself but I rather upstream this change so I don't have maintain a custom fork.

Is this something that you would be willing to merge? I guess this would add a couple of parameters to config with an additional motor error for exceeding the thermal limit.

samuelsadok, 2020-05-28:
Thanks for your interest!
Yes that's in general a feature we should have and would be willing to merge if someone looks into it.

There's already temperature monitoring for the on-board FETs in `Motor::update_thermal_limits()`. 
Here's how it works:

 - `motor.config.inverter_temp_limit_lower`: the FET temperature at which motor current limiting kicks in (up to this point the current limit is taken from `motor.config.current_lim`)
 - `motor.config.inverter_temp_limit_upper`: the FET temperature at which current is limited to 0A
 - `motor.config.inverter_temp_limit_upper + 5°C`: the hard limit at which the ODrive "panics" with `ERROR_INVERTER_OVER_TEMP` (this should ideally be configurable)

The motor temperature monitoring would probably look almost the same except that the sensor voltage current is retrieved from GPIOs (you can look at the sin/cos encoder implementation for this).

Going forward (not required for this) it would probably make sense to generalize the way we apply current limits to the FOC controller, so that current limits can be injected from anywhere in the system. Something like an array of pointers from which the FOC controller uses the smallest value.

rowanG077, 2020-05-28:
@samuelsadok Oke great! Will look at it next week when my thermistors have arrived!

rowanG077, 2020-05-28:
@samuelsadok The thermistor coefficient for the FETs thermistor are there to get a better readout then just applying a linear calculation?

https://github.com/madcowswe/ODrive/blob/5770b3d92d090826ef9b226b3ce44ed00f84deff/Firmware/MotorControl/board_config_v3.h#L68-L70

and used here:

https://github.com/madcowswe/ODrive/blob/767a2762f9b294b687d761029ef39e742bdf4539/Firmware/MotorControl/motor.cpp#L146-L150

Should it be possible for a user to manually specify these coefficient for their own thermistor with a maximum number of a degree 3 polynomial?

samuelsadok, 2020-05-28:
There are two common types of thermistors. One is Pt100 (or Pt1000), which has a roughly linear relationship between temperature and resistivity and the other ones are NTCs/PTCs, which [have an exponential relationship](https://en.wikipedia.org/wiki/Thermistor#B_or_%CE%B2_parameter_equation).

I'm not sure why we're using a polynomial fit instead of the exponential model, perhaps because `logf()` is much slower (@madcowswe?).

Here's a script to fit the polynomial based on the NTC's parameters: https://github.com/madcowswe/ODrive/blob/master/analysis/thermistors.py

Yes it would make sense to expose the coefficients as a user config. We could turn the script above into a convenience function in `odrivetool` so that users can directly set the coefficients from NTC parameters.

rowanG077, 2020-05-31:
Oke I will add this convenience function to the odrivetool as well. What's not clear to me is how I can migrate to a new config. Since I want to add new parameters the config stored in NVM no longer matches the struct definition.

samuelsadok, 2020-06-02:
If the length of the configuration in NVM and in the firmware mismatch, the firmware ignores the NVM. This is supposed to cover the most common use case, which is adding/removing config fields.
In that sense, the NVM is not persistent across firmware changes that touch the config format.

However if users upgrade through `odrivetool dfu`, then the odrivetool temporarily pulls all configuration values by name and then restores them by name in a best-effort fashion.

rowanG077, 2020-06-05:
I have started doing this and have a basic implementation ready. See my comment in the pull request itself. I still have some open points.

https://github.com/madcowswe/ODrive/pull/419

rowanG077, 2020-06-05:
There seems to be thermistor on the aux port but this isn't not used at all in the firmware? I can add this now quite easily in my pull request to introduce a current limit based on the thermistor if you guys want.

rowanG077, 2020-07-20:
This issue can be closed :)!

samuelsadok, 2020-07-20:
Yup, thanks for your work!

closed 2020-07-20 by samuelsadok