odriverobotics/ODrive #420: Stuck in DRV8301_readSpi infinite loop when using absolute encoders

jhurliman, 2020-06-06:
Here's my setup: ODrive 3.6-56V with AMT203-V absolute encoders running the latest firmware from `devel` (essentially the 0.5.0-rc). I was able to flash the new firmware, run the latest odrivetool, set the encoder CPR to 4096 and encoder mode to `SpiAbsCui` for axis0, then run motor calibration, encoder offset calibration, and go into closed loop control mode. Everything seemed great. However, when I powered the device off then powered it back on again, it refused to respond to requested_state commands. I flashed a debug version of the firmware and connected via ST-LINK, and found it was getting stuck on the following line in `drv8301.c`:

```
while((DRV8301_readSpi(handle,DRV8301_RegName_Status_1) & DRV8301_STATUS1_FAULT_BITS) != 0);
```

The value 0xFFFF was being returned each time, causing that check to fail.

If I set the encoder mode to 0 (Incremental) for both axes, save_configuration(), and reboot() I can get past that. Then, I can set the encoder mode back to 0x100 (SpiAbsCui) and set requested_state to AXIS_STATE_ENCODER_OFFSET_CALIBRATION. Then finally, closed loop control. But if I leave the encoder mode as SpiAbsCui and power cycle, I'm stuck back in that same startup loop.

samuelsadok, 2020-06-08:
For  `encoder.config.mode` to take effect you _need_ to reboot actually. Then, after the reboot, you need to run offset calibration once and save it. See [here](https://github.com/madcowswe/ODrive/blob/rc-v0.5.0/docs/encoders.md).

Do you have the incremental output of the encoder also connected to the ODrive? If so, I think what's happening is that your encoder is just being used as a normal incremental encoder until reboot. For an SPI encoder you don't need to connect the A/B/Z lines at all.

The other thing is, looking at the datasheet, the protocols for [CUI AMT20x](https://www.cuidevices.com/product/resource/amt20.pdf) and [CUI AMT23x](https://www.cuidevices.com/product/resource/amt23.pdf) seem incompatible. The latter one is implemented in `rc-v0.5.0` while the former is not. Feel free to open a feature request issue for the AM20x (or better yet a PR).

jhurliman, 2020-06-09:
We're going to order CUI AMT233B encoders and report back.

jhurliman, 2020-07-11:
The `AMT203-V` encoders have been swapped out with a pair of `CUI AMT232B-V` encoders. I'm seeing the same issue with `DRV8301_readSpi()` continuously reading `0xFFFF` causing an infinite loop.

jhurliman, 2020-07-11:
Oscilloscope for MISO with the encoder disconnected:

[image]

With the encoder connected:

[image]

We'll address this on the hardware side and report back. Thanks for the pointer.

samuelsadok, 2020-07-13:
Maybe nCS is not connected properly to the encoder and as a result the encoder is permanently enabled. If you measure the encoder's nCS you should see a steady 8kHz signal after the ODrive has started successfully. If you don't see this you have perhaps connected nCS to the wrong pin.

jhurliman, 2020-07-24:
Adding a pull-up resistor allowed us to successfully boot and go into idle state with the absolute encoders connected. However, there were various issues with calibration and even after hacking around them with unplugging and replugging we cannot get the device to go into closed loop control with absolute encoders connected and enabled. I tested #431 (merged into latest `devel`) to see if that improved the situation, but it seems to make things worse as we get stuck in the infinite loop both with and without the pull up resistor.

samuelsadok, 2020-07-27:
Where did you add the pull-up resistor? DId you check if the 8kHz signal appears on the encoder's nCS?
Additionally, there's a hack you can try: add a resistor (something like 100Ω...1kΩ) on MISO between the encoder and ODrive (in series).

wisitmax, 2020-07-27:
We added a 100k pull-up resistor between GPIO7 and 3.3V (did the same on GPIO8). After setting the encoder config and reboot, there weren't any 8kHz signal on GPIO7/8. But once we unplugged the encoder cable, the 8kHz appeared. Encoders appear to be working (pos_abs changes as the motor moves) as well after unplugging and replugging.

[image]

[image]

wisitmax, 2020-07-27:
Adding 1k res to the MISO line in series between encoder and ODrive fixes it.

wisitmax, 2020-07-28:
However, when we use two encoders (each has 1k resistor in series on its MISO line), we got ERROR_ABS_SPI_COM_FAIL on both axes.

Wetmelon, 2020-07-28:
Ah, I wish I'd seen this before.  I tested the AMT203 encoders and had this same problem.  They do technically function but I had all sorts of headaches.  IIRC I ended up putting a capacitor filter on CS but still regularly had problems getting the board to boot correctly.

Wetmelon, 2020-07-28:
I actually connected the shield ring terminal to ground, and it fixes everything :)
[image]

samuelsadok, 2020-07-28:
Lol so the SCLK signal induces ripples on nCS, causing spurious activations/deactivations of the AMT203's MISO output? From just looking at the nCS signal there I wouldn't have guess that those ripples cause confusion at the encoder side. Is this measured close to the ODrive or close to the encoder?

wisitmax, 2020-07-28:
I think I found the problem but I'm not sure how to fix it. As soon as I connect the 2nd MISO from the 2nd encoder, the MISO line (dark blue) got cut in about half.

yellow: nCS0
cyan: SCK
pink: nCS1
dark blue: MISO

With one encoder.
[image]

With two encoders. MISO line (dark blue) got cut in half, LOW signal is ~2V.
[image]

wisitmax, 2020-07-30:
It seems that the encoders are constantly talking on the MISO line. The presence of more than one would pull the MISO line high causing errors. We added a tri-state buffer gate (e.g. 74AHC1G125) on each encoder's MISO line and that fixed it.

Wetmelon, 2020-07-31:
Yeah, something like this?  
[image]

madcowswe, 2020-07-31:
I had a look at the datasheets, and indeed the CUI AMT203 uses SPI, the AMT232 actually claims to be SSI.
However, it's not quite standard SSI. Since they still have a "chip select" to frame the data, it seems more like SPI but without tristate. Strange choice, since it's not actually compatible with SSI.

Good thing ODrive v4 is single axis.

Wetmelon, 2020-11-10:
We sent this information to CUI, and they confirmed there is a bug in the PCB and will be fixing it such that their SSI mode actually matches the datasheet. https://www.cuidevices.com/blog/integrating-absolute-encoders-an-overview-of-spi-rs-485-and-ssi-protocols#synchronous-serial-interface-ssi-

Additionally, `devel` branch will no longer block on DRV failure, but instead throw a DRV FAULT and proceed: 1d0fcc5fcffae2873ff8fbb499a5187c10dc11be

closed 2020-11-10 by Wetmelon

wernerdaehn, 2020-11-24:
Just for confirmation, what is the summary of the problem? I'd like to replace the CUI encoder with an absolute encoder of same shape/dimensions. 
Am I correct in the assumption that with one encoder all is fine, the issue is when two encoders are connected?
And what will be the solution? CUI will change something or oDrive PCB?

Wetmelon, 2020-11-25:
The MISO line doesn't tri-state properly.  You can just put a tri-state buffer on the line.  CUI is changing something, and we're coming out with an "Isolator" board that will effectively do the same thing

joshmouch, 2023-09-30:
Is this fixed yet?

Wetmelon, 2023-09-30:
> Is this fixed yet?

No, this is a hardware problem. The AMT203 needs a tri-state buffer on the MISO line or it will respond to SPI clock signals during DRV configuration at startup.

joshmouch, 2023-09-30:
In your last post you said "CUI is changing something", so I thought that meant the hardware change.

Wetmelon, 2023-09-30:
> In your last post you said "CUI is changing something", so I thought that meant the hardware change.

Oh, no idea.  CUI never got back to us and we've moved onto other problems.