odriverobotics/ODrive #422: Axis number: constexpr size_t AXIS_COUNT = 1?

zhongmu, 2020-06-08:
I plan to add a temperature sensor (i2c) into odrive. So i use M0 with default coonfiguration, and reconfigure firmware of M1. But odrivetool can not connect to the MCU with AXIS_COUNT = 1.
How can i just initialize one motor?

samuelsadok, 2020-06-08:
Why do you need to disable M1 to add a I2C temperature sensor?

Currently the AXIS_COUNT constant exists only for readability, not for configuration. Several parts of the firmware still assume a count of 2, regardless of this variable. You can search for `axes[1]` as a start. In addition, see https://github.com/madcowswe/ODrive/pull/292#issuecomment-638847812.

zhongmu, 2020-06-08:
Thank you very much!
I want to measure windings temperature of M0 and then tune controllers because resistance and inductance varies with temperature. This is a test for another project.

zhongmu, 2020-06-08:
Sorry!
I just found PB8 and PB9 can serve as I2C ports.

zhongmu, 2020-06-08:
Emmm, PB8 and PB9 are initialized as I2C_1 ports in HAL_I2C_MspInit. Maybe I can modify less by replacing them with PB6 and PB7. But what will happen next, i am not sure. I need to learn more about the code.

samuelsadok, 2020-06-08:
That could work. You'd need un-use them in `HAL_TIM_Encoder_MspInit` to prevent collision. Also beware that currently I2C is configured as slave, not as master.

closed 2020-08-31 by samuelsadok