odriverobotics/ODrive #425: Implement temperature reading for AUX thermistor

rowanG077, 2020-06-11:
When implement motor thermistor support I noticed there is a thermistor at the AUX FETs that is not used in the code (or maybe I'm blind). It would be great if we start reading out this thermistor as well.

samuelsadok, 2020-06-11:
Yes that thermistor is unused. You can add code to read out this thermistor if you want but when it comes to acting on it things become a bit tricky.

We can't simply reduce brake resistor current if the AUX fets get too hot because that would mean that excess power is sunk into the power supply (which can destroy the PS or the ODrive or both).
Instead we'd need to provide the motor(s) with feedback that they need to reduce their negative current but not their positive current.

Before we add too much complex special purpose code I'd prefer to think about something like a generalized power solver which attempts to satisfy a prioritized list of constraints of all involved components (motors, brake resistor, power supply).

Maybe @madcowswe has an opinion on this.

Wetmelon, 2021-09-13:
Hi @rowanG077, we're going to harden the v3 firmware as-is without this feature.  We always welcome well-tested PRs.

closed 2021-09-13 by Wetmelon