odriverobotics/ODrive #428: UART communication from Arduino to ODrive delay

Peter-haotian-cui, 2020-06-20:
Hi, 
I'm using a stm32duino to communicate with ODrive and found a large delay to get a response (almost 1ms). The stm32 UART baudrate is set to 1Mbps, and using a self build ODrive with the same baudrate. Because I want to have a high frequency control (>500Hz), this 1ms delay is too much for my application. Does anyone knows about the reason and how to reduce the response delay.
[image]

Thanks!

samuelsadok, 2020-07-08:
That might be because of this delay: https://github.com/madcowswe/ODrive/blob/master/Firmware/communication/interface_uart.cpp#L89

This delay exists because otherwise the loop would be a busy spin and starve other threads (such as motor control). There's a PR #393 which changes the UART code to use the RX interrupt instead of polling, maybe this fixes your issue.

samuelsadok, 2020-08-31:
I'm gonna close this now that we have an increased polling frequency but if it's still an issue feel free to reopen.

closed 2020-08-31 by samuelsadok

EMHacker-alt, 2020-08-31:
This is great !!!