odriverobotics/ODrive #429: Encoder::abs_spi_cb()

zytra, 2020-06-22:
Hi, after chatting with @madcowswe on discord I summarized on my Discourse thread; what's following should be a relatively straight forward 'enhancement' (more of a future proof change).

Inside Encoder::abs_spi_cb() pos is a variable used to pass the rawdata from the Rx register to pos_abs_ which is a global variable that contains the encoder position.

pos is a uint16_t [declared in Encoder::abs_spi_cb()]
pos_abs_ is a uint32_t [declared in encoder.hpp]

I'm working on adding a new "encoder" type that is 24 bit. In order to keep the added code as uninvasive as possible changing pos type to a uint32_t would be ideal. I have made that change locally.

FYI, I'm basing all my work off the RC 0.5.0

Note: here's the discourse thread: https://discourse.odriverobotics.com/t/adding-support-for-another-spi-absolute-encoder/4799/25?u=zytra

Wetmelon, 2020-07-28:
Did you make a pull request for this already?

zytra, 2020-07-28:
no I did not, but this one is nothing more than changing uint16_t to uint32_t .
nothing more to it as far as I can tell

samuelsadok, 2020-07-31:
IMO this is something we should do at the same time as adding support for a 24-bit encoder. But by that time we might have separate classes for each SPI encoder protocol anyway so each `abs_spi_cb()` function would contain only code tailored to that encoder.

closed 2020-08-31 by samuelsadok