odriverobotics/ODrive #434: Trapezoidal control looses force on engaging

dronus, 2020-07-03:
If the control is running on high error (eg. closed loop control without integral, the motor won't reach target position) and trapezoidal movement is engaged via the move_to_pos command, the force drops and the motor gives way to external forces, before finally engaging again.

It just feels like the trajectory is planned by the current position and not the setpoint, although the code seems otherwise.

With setpoint control, by programming a target that can't be reached, the force can be controlled by choosing how far the setpoint is behind an obstacle.  However if using move_to_pos, the motor would be pushed back and then builds up force again  on any call of move_to_pos, even if  the programmed target would result in a higher final force.

Why is this, and how can it be prevented?

Wetmelon, 2020-07-28:
move_to_pos is a very simplistic planner that uses the current position and velocity (assuming they're achievable) as the starting point of the control.  It is intended for _unrestricted_ point-to-point movement.  It is not intended for your use case, sorry :/

dronus, 2020-10-13:
I thought it uses the current controller.pos_setpoint as the starting point? Code reads like this.
But the real movement feels like it starts at encoder.pos_estimate, even if pos_setpoint is already far off. That is what I am wondering about!

Wetmelon, 2020-10-20:
> I thought it uses the current controller.pos_setpoint as the starting point?

Yes, and if you're operating the drive within its limits in Trapezoidal mode, then `pos_estimate` is approximately == `pos_setpoint`.  I'll look into it, it's possible the integrator is resetting for some reason?

closed 2021-09-13 by Wetmelon