odriverobotics/ODrive #435: Add support for encoder memory.

rowanG077, 2020-07-08:
Many applications have requirements where startup calibration can't be done due to various reasons, see various posts on discourse.

I want to add support for externally polling an SPI IC to get the absolute count during calibration instead. The [LS7366](https://lsicsi.com/datasheets/LS7366R.pdf) is a very simple IC that can function for this purpose as long as it's backed by a battery.

I would like to implement this if you would allow support for it.

samuelsadok, 2020-07-10:
A small number of absolute SPI encoders are already implemented on the `rc-v0.5.0` and `devel` branches, see here: https://github.com/madcowswe/ODrive/blob/rc-v0.5.0/docs/encoders.md#spi-encoders

Does this cover what you need? If not you could implement the LS7366 protocol as an additional encoder mode (based on the `devel` branch).

rowanG077, 2020-07-10:
I assumed(maybe wrongly) a raw quadrature signal is better during control due to the latency that is incurred when asking a SPI device for the position of the motor. So the goal would be use SPI only once during setup to get the current absolute position and then use the normal quadrature signals during control.

samuelsadok, 2020-07-13:
I would approach this on two levels actually: one is the pure protocol implementation of LS7366 (should be fairly straightforward) and the other one is a fusion of data from two encoders. For the latter there are several valid reasons. For instance having one magnetic encoder for a rough absolute estimation (for commutation) and one incremental encoder for precise position control. Or one encoder before the gearbox (for commutation) and one after the gearbox (for position control that takes backlash into account).

Your use case is a special case of fusion and could admittedly be implemented more easily (as you described). However unless the edge in latency is something you need right now I would wait until we have the encoder code more cleaned up so there is a clearer path to sensor fusion.

samuelsadok, 2020-08-31:
I filed #470 to track encoder fusion so I'll close this issue.

closed 2020-08-31 by samuelsadok