odriverobotics/ODrive #438: USB responses over 64 bytes are silently dropped or cause buffer over-read

durka, 2020-07-13:
This bug has multiple layers which hide each other, but I am fairly confident it's a real bug.

A loop in `interface_usb.cpp` attempts to send everything passed to it, even if the passed-in buffer is larger than a USB packet:
https://github.com/madcowswe/ODrive/blob/306000f46e5d04f21177ee05fcd11469e5282295/Firmware/communication/interface_usb.cpp#L60-L71

However, because the original `length` is passed through on line 63 instead of the newly calculated `chunk`, the loop fails in its stated purpose.

The inner type, `USBSender`, simply returns -1 if asked to send more than 64 bytes, so the loop would exit on the first iteration if `length > 64`. (If `length` is precisely 64, we get through, but I think that's not really intended and the comparison on line 62 ought to be `<=`.)

As it is, the `respond` function in `ascii_protocol.cpp` tries not to send more than 64 bytes anyway, so the bug probably won't come up. But, that's not airtight for two reasons:
1. The buffer in `respond` is declared as size 64, but not using the `USB_TX_DATA_SIZE` definition, so could easily get out of sync with that
2. I say *tried* because `respond` uses `snprintf` incorrectly. From the man page:
  > The functions snprintf() and vsnprintf() do not write more than size bytes (including the terminating null byte ('\0')).  If the output was truncated due to this limit, then the return value is the number  of  characters  (excluding  the  terminating null byte) which would have been written to the final string if enough space had been available.  Thus, a return value of size or more means that the output was truncated.

So if one of the `snprintf`s in `respond` *would* write more than 64 bytes, it does not, but `len` is incremented anyway. Then a subsequent `snprintf` or the USB sending code will read past the end of the buffer.

madcowswe, 2020-07-14:
Thanks for the detailed bug report!

samuelsadok, 2020-07-20:
Thanks for the well-structured bug report!

These are valid concerns and I have addressed them in #441. I believe there is, in released firmware, no actual way to trigger them (the longest response line is 50 characters) but dormant bugs are still bad of course. While looking into this I actually came across another previously dormant bug that awoke just recently (see PR).

Feel free to take a look and comment, otherwise I'll go on merging it in a bit.

> If length is precisely 64, we get through, but I think that's not really intended and the comparison on line 62 ought to be <=.

Here I'm not sure what you mean. Note that the expressions `length < USB_TX_DATA_SIZE ? length : USB_TX_DATA_SIZE` and `length <= USB_TX_DATA_SIZE ? length : USB_TX_DATA_SIZE` are equivalent. A chunk size of exactly 64 bytes should be ok for the lower layers.

closed 2020-08-29 by madcowswe