odriverobotics/ODrive #444: M0 current loop noise/unstable

EMHacker-alt, 2020-07-24:
I just brought 6 ODrive v3.6, and found out 2 of them M0 current loop are abnormal. And the other boards, current loop also has too much current noise compared to M1. The M0 motor either make high frequency noise or vibration. I tested this only under current mode, so it is no possible that velocity or position loop affect this. And I tried to set the current i_gain to zero, only left the p_gain, but still the same. Further more, I reconnect the same motor and encoder to M1, no problem at all. This is weird, it seems like the current loop is unstable in high gain or some unknown lag caused this? 
Below is a waveform snapshot which shows the phase C PWM, current test point and the current waveform for a external isolated current probe.
[image]

madcowswe, 2020-07-31:
Hi,
Are these ODrives official ones from odriverobotics.com/shop?
What kind of noise are you getting, do you have a video or other data showing it?
Assuming you are rotating the motor in the capture shown, this result looks normal to me.
What motor are you connected to?

madcowswe, 2020-08-04:
Thanks for the video, this helps me recognize the sound.
I think indeed the current controller is a bit overtuned and getting a bit unstable. To be sure, can you check that your encoder is correctly reading 0 change in position in this test condition.
Assuming encoder is working fine: I suspect that the reason this is happening is because your motor inductance is quite high compared to most motors we have tested on. The current controller gains are derived from the specified bandwidth and the motor inductance. Can you try to set `motor.config.current_control_bandwidth = 500` or maybe even try 200 and see how that goes?

EMHacker-alt, 2020-08-05:
Thanks for your advices! Changing the `current_control_bandwidth` to 200Hz greatly improves the problem. While, I also can see that motion becomes slower than before. So I think higher current bandwidth is necessary. And I still curious about why Motor 1 has no problem at all even with 1000Hz current bandwidth? In addition, in order to eliminate the affect of encoder, I checked the AB pin of encoder, no pulses appears.

tombsar, 2020-08-26:
Same issue here. M0 emits a loud hissing noise in closed loop control. Official ODrive 3.6, official 5065 motor, default everything. M1 creates the same type of noise, but much quieter.

Wetmelon, 2021-09-13:
Does this noise change depending on the firmware version loaded on the board?

Wetmelon, 2021-09-13:
After reviewing the whole conversation, I'm going to close this with the workaround to use a higher current control bandwidth.

closed 2021-09-13 by Wetmelon