odriverobotics/ODrive #450: Dangerous state after canceling trapezoidal control

dronus, 2020-08-12:
If a trapezoidal motion is interrupted by setting axis state to AXIS_STATE_IDLE (eg. some kind of emergency stop) resuming operation is dangerous.

I tried both
`
axis.controller.move_to_pos(pos)
axis.requested_state=AXIS_STATE_CLOSED_LOOP_CONTROL
`
and 
`
axis.requested_state=AXIS_STATE_CLOSED_LOOP_CONTROL
axis.controller.move_to_pos(pos)
`
but there is always a massive non-trapezoidal motion. Looks like the old motion was still dry-running in IDLE state and the new one is taking over a position that does not match reality. 

I've also tried adding `axis.controller.pos_setpoint=axis.encoder.pos_estimate` first, as there is some position error after interrupting the control of course, but there is still some non-trapezoidal motion.

Is there any safe procedure for recovering?

dronus, 2020-08-13:
Seems the following sequence will mitgate the problem:

    axis.controller.pos_setpoint=axis.encoder.pos_estimate
    axis.controller.move_to_pos(axis.encoder.pos_estimate)
    axis.requested_state=AXIS_STATE_CLOSED_LOOP_CONTROL
    axis.controller.move_to_pos(pos)

It seems that the trajectory is always planned starting at `pos_setpoint`, not the real world state. Is this on purpose?

Wetmelon, 2020-08-14:
What version is this on?

dronus, 2020-08-16:
I think the code line in doubt is https://github.com/madcowswe/ODrive/blob/ffbeaed61b74cb23c522681b062a0337c601330d/Firmware/MotorControl/controller.cpp#L53
which is still in master. planTrapezoidal initializes on pos_setpoint_, which may be far away from encoder pos_estimate after idling.

Wetmelon, 2020-08-19:
Yeah, this is taken care of in `devel`, it initializes pos_setpoint = pos_estimate upon entering closed loop.

closed 2020-08-19 by Wetmelon