odriverobotics/ODrive #452: Endstop pullup configuration crushed by GPIO initialisation

TobinHall, 2020-08-23:
In the devel branch the endstop configuration allows for the internal pull-up to be enabled.
The endstop code correctly sets the mode and pull-up state but this happens before the main function initialises every GPIO pin.
There is no way to set a pull-up with the current GPIO modes so it is currently not possible to use an internal pull-up with endstops.

Wetmelon, 2020-08-24:
Ah.  Oops.

Wetmelon, 2020-08-24:
Thanks for the heads up.  We're doing a lot of low level refactoring on devel right now so it's not very stable, recommend using rc-v0.5.1 in the meantime.

samuelsadok, 2020-08-24:
Thanks for reporting, I overlooked this one. My proposed fix is to move the responsibility of initializing the pullup/pulldown mode to the `odrv.config.gpioX_mode` variable. Here's a corresponding PR: #453.

TobinHall, 2020-08-24:
> Thanks for reporting, I overlooked this one. My proposed fix is to move the responsibility of initializing the pullup/pulldown mode to the `odrv.config.gpioX_mode` variable. Here's a corresponding PR: #453.

Nice, this works well in my setup. I believe that there is now a redundant pullup variable in the endstop config.

Is it worth also having a mode for encoder nCS lines? A push-pull output defaulted high. This might help with one of the causes of #420 if external pull-ups aren't used.

samuelsadok, 2020-08-25:
Thanks, I removed the variable.

Encoder nCS are already [configured](https://github.com/madcowswe/ODrive/blob/devel/Firmware/MotorControl/encoder.cpp#L456-L463) as push-pull defaulting to high. But admittedly this is not the best place to do this and https://github.com/madcowswe/ODrive/pull/451#issuecomment-679843656 brings up a related problem.

So going back to the scratchpad:
 - Inputs can be sampled by multiple components at the same time but they can only have one configuration. So it makes sense to have a general purpose input mode `GPIO_MODE_DIGITAL_IN` and have variants of it `GPIO_MODE_DIGITAL_IN_PULL_UP/DOWN` (could also have a separate pullup/down variable per GPIO but this was less intrusive).
 - Outputs can only be driven by one component at a time. So it makes sense to directly assign the GPIO to the corresponding component. I think it's sufficient to just set the high level component (`GPIO_MODE_SPI0 ` rather than `GPIO_MODE_SPI0_CS`) and handle the distinction between nCS and hardware SPI pins automatically in firmware. This way we avoid having a plethora of modes but still give the user a quick overview what uses which pin.

As a side note you could ask why we added these modes in the first place instead of just letting each component handle GPIO initialization itself. The reason is that that way it would be very hard for a user or GUI to quickly tell "ah I need to disable UART and PWM to use GPIO1 as nCS". This is especially a concern with a constantly changing firmware.

TobinHall, 2020-08-25:
Fixed by #453.

closed 2020-08-25 by TobinHall