odriverobotics/ODrive #456: odrivetool dfu fails to find firmware releases from GitHub

tombsar, 2020-08-27:
I'm using odrivetool from a Raspberry Pi (installed with pip) to update the firmware on an ODrive 3.6 24V. This is what happens:

```
$ odrivetool dfu
ODrive control utility v0.4.12
Waiting for ODrive...
Found ODrive 2080396B4D4D (v3.6-24V) with firmware [unknown version] in DFU mode
Checking online for newest firmware...'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
'_sre.SRE_Match' object has no attribute '__getitem__'
Traceback (most recent call last):
  File "/usr/local/bin/odrivetool", line 142, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 454, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 313, in update_device
    raise Exception("could not find any firmware release for this board version")
Exception: could not find any firmware release for this board version
```

This looks like a mistake in the Python code rather than anything related to the hardware. I think something is going wrong in this function: https://github.com/madcowswe/ODrive/blob/master/tools/odrive/dfu.py#L133

tombsar, 2020-08-27:
Looks like this is a Python 2 vs 3 compatibility issue. In Python 2 you must use `.group(i)` instead of `[i]` here: https://github.com/madcowswe/ODrive/blob/master/tools/odrive/dfu.py#L145

samuelsadok, 2020-08-31:
I think we actually never tested DFU script on Python 2. Thanks for your suggested fix. I pushed it together with a few more fixes here: https://github.com/madcowswe/ODrive/commit/96cbe69f8aaabba669e39331e15ca364dd4e6eba. With this I was able to run a firmware upgrade on Python 2.

Anyway I highly recommend using Python 3 instead (install odrivetool with `pip3` rather than `pip`). The only reason we still remain compatible with Python 2 is that some ODrive users are afflicted with gravely outdated systems.

tombsar, 2020-08-31:
Fantastic; thanks!

I only used Python 2 because I encountered errors when first trying to install with `pip3` (problems with my environment, not odrivetool; fixed by uninstalling Python 2 versions of some of the dependencies).

closed 2020-08-31 by tombsar