odriverobotics/ODrive #457: utils.yes_no_prompt() does not handle an empty response

tombsar, 2020-08-27:
The function is supposed to return a default value if the user doesn't supply anything, however it instead raises an exception:
```
Traceback (most recent call last):
  File "/usr/local/bin/odrivetool", line 142, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 454, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 344, in update_device
    elif not odrive.utils.yes_no_prompt("The configuration cannot be backed up because the device is already in DFU mode. The configuration may be lost after updating. Do you want to continue anyway?", True):
  File "/usr/local/lib/python2.7/dist-packages/odrive/utils.py", line 206, in yes_no_prompt
    choice = input().lower()
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing
```

Possibly a Python 2 vs 3 compatibility issue.

tombsar, 2020-08-27:
Actually, it seems to be completely broken. Even if I type a response I get this:
```
Traceback (most recent call last):
  File "/usr/local/bin/odrivetool", line 142, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 454, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/usr/local/lib/python2.7/dist-packages/odrive/dfu.py", line 344, in update_device
    elif not odrive.utils.yes_no_prompt("The configuration cannot be backed up because the device is already in DFU mode. The configuration may be lost after updating. Do you want to continue anyway?", True):
  File "/usr/local/lib/python2.7/dist-packages/odrive/utils.py", line 206, in yes_no_prompt
    choice = input().lower()
  File "<string>", line 1, in <module>
NameError: name 'Y' is not defined
```

tombsar, 2020-08-27:
Looks like I needed to type `'y'` with quotes. Is that intended behaviour? If so, I think the prompt needs to be updated to show that.

samuelsadok, 2020-08-31:
Nope this is not intended behavior. It seems the semantics of `input()` are not the same in Python 2 and 3. As such it's another aspect of #456 and fixed in 96cbe69f8aaabba669e39331e15ca364dd4e6eba.

closed 2020-08-31 by samuelsadok