odriverobotics/ODrive #460: Setting `watchdog_timeout` from ASCII protocol does not start the watchdog timer

jdrnd, 2020-08-28:
Steps to reproduce:
1. Set up motors to boot into velocity control mode with the watchdog timer disabled.
2. Set the watchdog timer over USB ASCII for `x` seconds via `w axis0.config.watchdog_timeout x`
3. Note that the property `axis0.config.watchdog_timeout` is set to your commanded value
4. Watchdog timer never triggers

If using ODrivetool via USB, the watchdog timer starts and triggers as expected. Note on this, in order to restart the watchdog timer I had to reset `watchdog_timeout` to zero, clear the error, and reset watchdog timer to the desired value, when I expected that clearing the error would be enough.

jdrnd, 2020-08-28:
It looks like this issue and a potential fix were previously referenced at https://discourse.odriverobotics.com/t/communication-watchdog/1841/13

samuelsadok, 2020-08-31:
Thanks for the report! I take it this issue occurs on the `master` branch (or 0.4.12 release)? If so then yes this was a known issue. However the back end implementation of the ASCII protocol changed recently. So on `devel` as well as the [next release candidate](https://github.com/madcowswe/ODrive/releases/tag/v0.5.1-rc2) it should be fixed. Please let me know if you still get issues with the new firmware.

closed 2021-09-13 by Wetmelon