odriverobotics/ODrive #470: Encoder fusion

samuelsadok, 2020-08-31:
There are at least two legitimate use cases for fusing several position and/or velocity estimates:

 - Detecting incremental encoder drift / disconnect by comparing to sensorless estimator.
 - Augmenting precise incremental encoders with imprecise but absolute magnetic encoders for calibrationless startup.

dronus, 2020-10-13:
Another idea: Just allow for a host correction, eg. a softly applied encoder.set_linear_count() , which alows glitchless updates (even when moving on trapezoidal control).  This simple patch would allow the host to feed data from absolute or  other position cues to the ODrive while moving, allowing for small corrections over time. This also enables the use of non-synchronized mechanics which show slip, eg. wheels, with external position feedback.

ddieffen, 2020-10-29:
Some commercial drives I use have that feature. It is a special mode where at low speed, the encoders are used, and above a about 20% of the speed, when the backEMF is good enough to be measures, the controller applies more weigth to the angle estimator using the backEMF.
The transition is smooths and allows to use the best of both cases : encorder at slow speeds, backEMF when speed is large enough.