odriverobotics/ODrive #473: First ASCII command recieved after reboot replies "unknown command"

jdrnd, 2020-08-31:
When using v0.5.1-rc2, after reboot the first ASCII command sent to the device will receive a reply "unknown command". Subsequent commands sent will behave as expected. Tested with `w`, `f`, and `r` command types.

samuelsadok, 2020-09-01:
How are you testing this? Are you using the ASCII protocol over UART or over the USB-CDC interface? I could not reproduce the issue so far.

I tried the following on a Raspberry Pi:
 1. `wget https://github.com/madcowswe/ODrive/releases/download/v0.5.1-rc2/ODriveFirmware-v3.6-56V.hex`
 2. `odrivetool dfu ODriveFirmware-v3.6-56V.elf` (I used odrivetool from the repository but that doesn't make a difference here)
 3. Open `screen /dev/ttyS0 115200` in one terminal. Send a command to make sure the connection is good.
 4. Open odrivetool in another terminal.
 5. In odrivetool run `odrv0.reboot()` (now the ODrive prints a stray character on UART which is not intended but it's not the issue you're describing)
 6. In `screen` type `r vbus_voltage` <kbd>Enter</kbd>

This returned the vbus voltage as it ought to.

Note that control keys are not handled by the ODrive, so things like typing `r vbus_voltagw` <kbd>Backspace</kbd> `e` <kbd>Enter</kbd> will not work.

Wetmelon, 2021-09-13:
OP never responded.  Closing.

closed 2021-09-13 by Wetmelon