odriverobotics/ODrive #476: USB hard deadlock

dronus, 2020-09-01:
My Odrive USB interface just locked up:

-The python remote object shows emptied out ( no methods / properties )
-Despite this, closed-loop motor control is still engaged and seems fine
-The /dev/ttyACM... device is still existent, eg. USB device did not disconnect by itself.
-Trying to reconnect by odrive.find_any(serial_number=...) does hang forever
-If USB is unplugged and plugged again, dmesg shows the errors:

    [  902.948653] usb 1-1.2: new full-speed USB device number 9 using xhci_hcd
    [  903.048858] usb 1-1.2: device descriptor read/64, error -32

or later

    [  905.128646] usb 1-1.2: Device not responding to setup address.

-Rebooting the computer does not change anything
-Power cycling the ODrive solves the problem.

So it seems to be some hard lockup in the low level USB protocol stack. 

Maybe some watchdog to reset USB on the device side is possible?

Wetmelon, 2020-09-05:
Rebooting the ODrive should reset all internal state.  Something on your PC side?  Maybe you can remove & reinstall the USB drivers (idk how Linux works)

dronus, 2020-09-06:
> Rebooting the ODrive should reset all internal state. 
Doesn't look so in this case! The only solution was power cycling on ODrive, which strongly suggest some state is kept by the USB interface device.

>Something on your PC side?
> Maybe you can remove & reinstall the USB drivers (idk how Linux works)
Drivers in Linux like this are always installed (they do not install on plugging the device like in Windows) and do not hold any state on the disk (unlike in Windows) so they basically can't break. They may hang of course, but reboot fixes that where in this case it doesn't.

samuelsadok, 2020-09-08:
It seems that the ODrive just stopped handling USB interrupts. The `dmesg` log hints at this by saying that the driver couldn't even read the USB descriptors. No response is different from a disconnect or device-side reset, which is signaled by letting D+ and D- go low. That explains why the device didn't disappear in `lsusb` and at `/dev/ttyACMx`.

Do you remember what you did to get to this state?

An upcoming PR will actually refactor the ODrive-specific part of the USB code so there's a non-zero chance that it's fixed on there.

dronus, 2020-09-08:
> It seems that the ODrive just stopped handling USB interrupts. The `dmesg` log hints at this by saying that the driver couldn't even read the USB descriptors. No response is different from a [...]

The ODrive in question had some heavy traffic (repeated phases of tight property read and write for trajectory control) and of course may have flaky USB wire or flaky power on the host (but not on the drive, it is powered at 36V and never reports voltage drops).

> 
> An upcoming PR will actually refactor the ODrive-specific part of the USB code so there's a non-zero chance that it's fixed on there.

Sounds good.

I will investigate that further, however I've only encountered it twice in about a month or so.  Still may be a showstopper for 24/7 unattended application, as it can't be recovered from the host side without adding an adddional power switch relay or something.

samuelsadok, 2020-09-08:
> Still may be a showstopper for 24/7 unattended application

Yes true. I think besides trying to fix this particular issue the solution to a wider range of problems that could prevent unattended operation (hard faults, program counter runaway, etc) is to make use of the hardware watchdog.

Wetmelon, 2021-09-13:
Closing for now, as the updates have done a lot of refactoring. Reopen if it's still an issue, please.

closed 2021-09-13 by Wetmelon