odriverobotics/ODrive #478: AS5048A encoder issue

Zhaotianzheng, 2020-09-02:
1.When I am using the SPI, I find that it will make drv8301 can not be configured and hault in motor configuration stage. I am currently make the cs pin to be high before moter configuration but I don't think this is a good idea. So I want to know if there is a better way to use SPI?
2. After it can use, I find I have to configure the cpr to 2**14 - 1 to make it work, otherwise it will give a strange vel_esitimate and get the overspeed error. So I think maybe the cpr should be 2**14 -1 or just something I did is not right.

Thank you for your helping.

ionelanton, 2020-09-02:
Hi!
I have the same problem. When I run requested_state = AXIS_STATE_FULL_CALIBRATION_SEQUENCE nothing happen.
I followed the instructions here: https://github.com/madcowswe/ODrive/blob/rc-v0.5.1/docs/encoders.md

How do you make the CS pin to be high?

Zhaotianzheng, 2020-09-02:
In the Axis::setup() function I add the CS pin configuration code to make it to high. Just configure the GPIO and write high to it.

PAJohnson, 2020-09-02:
Hi Zhaotianzheng. Which firmware branch are you using? Devel, Master, rc-v0.5.1?

Wetmelon, 2020-09-03:
A hotfix has been applied to `devel` and `rc-v0.5.1` branches.  There is a setup procedure required:

- Disconnect absolute encoders
- Configure encoders in odrivetool with correct CS pins & mode
- Save config
- Turn off power to the ODrive
- Plug in absolute encoders
- Turn power on

This should only be necessary for the first time, because we don't drive the GPIOs high unless the axis is configured for absolute SPI.

Zhaotianzheng, 2020-09-03:
> Hi Zhaotianzheng. Which firmware branch are you using? Devel, Master, rc-v0.5.1?

I used the old version, now with the rc-v0.5.1 I have no problem. Thanks.

Zhaotianzheng, 2020-09-03:
> A hotfix has been applied to devel and rc-v0.5.1 branches.  There is a setup procedure required:
>
> Disconnect absolute encoders
> Configure encoders in odrivetool with correct CS pins & mode
> Save [...]

Thank you, I didn't use the rc-v0.5.1. Now I have no lock up problem with the new version.