odriverobotics/ODrive #493: Odrive not obeying vel_setpoint set to stop?

Patronics, 2020-09-27:
As far as I can te

Wetmelon, 2020-09-28:
Hi Patronics, do you have anything set for current_setpoint?

Patronics, 2020-09-29:
I believe it was set to zero as well, but I'd have to double check. How should that effect things? I thought that would only be used if control_mode was set CTRL_MODE_CURRENT_CONTROL? Thanks for any help!

Patronics, 2020-09-29:
I just ran `odrivetool dfu` and updated the firmware (apparently from 4.11-dev to 4.12), and it seems like that may have fixed the issue. I still need to test a bit more with the control method I've been using, but this issue might have been resolved by the firmware update :)

Wetmelon, 2020-09-30:
Oh my yes, 4.11 is quite old.  We just released 5.1, which should get picked up by `odrivetool dfu` soon..

closed 2020-09-30 by Wetmelon