odriverobotics/ODrive #494: USB data line powered while disconnected

dronus, 2020-09-28:
When using several ODrives on an Raspberry Pi 4, the raspberry fails to boot. I've found some hints from other people mentioning USB problems, and it seems that cheap USB devices tend to pull the USB data lines when powered externally, back-powering the raspberry by the data lines and leading to a brownout condition before the raspberry supply sport enough power. 

So I've checked my ODrive, and in fact it does power D+ (testpoint TP2) prior to being plugged. 

So I guess this behaviour is causing the Raspberry to behave eratically. A single device doesn't seem to provide enough power, but multiple powered D+ lines finally powers the raspi power controller IC which then locks into brownout condition, preventing the raspberry from booting.

Is this just a software issue in the USB stack, or does this need a hardware fix?

samuelsadok, 2020-09-30:
It seems indeed that ODrive v3 is non-compliant to USB specs.

If I understand correctly, a compliant peripheral should monitor V_bus of the USB connector to know when to pull up D+. This pull-up is required to signal presence to the host.

The V_bus pin of the USB receptacle is not connected on ODrive v3 so we can't properly fix this in software. Maybe it's possible to work around this issue by floating D+ (PA12) most of the time and only once in a while (let's say for 100ms every second) switching the pin to USB mode.

Another (much simpler) workaround could be to connect the ODrive through a hub.

dronus, 2020-10-13:
By using a hub for some of the drives, the bug is indeed remedied. Question is if that would add noticeable latency.

samuelsadok, 2020-10-14:
While there could be an increased latency in theory I don't think it's noticeable in practice.

> The Transaction Translator takes high-speed split transactions and translates them to full-/low-speed transactions
when the hub is operating at high-speed and has full-/low-speed devices attached.

—USB Specification 2.0, Section 11.1.1

The Hub-to-ODrive connection operates at Full Speed (12Mb/s) while the Host-to-Hub connection operates at High Speed (480Mb/s) or even Super Speed (1Gb/s). The faster baudrates also come with shorter transmission time frames which also helps with latency. So my guess is that the added latency is negligible compared to the inherent bus-time-quantization-caused latency of the Host/Hub-to-ODrive connection.

Anyway, if you have hard real time constraints you may want to consider using a [CAN shield](https://www.waveshare.com/rs485-can-hat.htm) on the Raspberry Pi.

dronus, 2020-10-24:
A partially fix for the beginning would be (if possible by software) to delay the USB initalisation by some part of a second on power on, to allow the host pass it's power check if both devices are powered at almost the same time (eg. sharing the same supply). 

Not sure how long this delay on power on is currently?

samuelsadok, 2020-10-26:
Currently there's no deliberate delay before USB is enabled. So I'd expect it to be on the order of tens of milliseconds.
If you want to try it out you can add `osDelay(some_number_in_milliseconds);` before `MX_USB_DEVICE_Init();` in main.cpp (on the `devel` branch).

closed 2021-09-13 by Wetmelon