odriverobotics/ODrive #495: parameter calibration_current without any effect

sl13, 2020-09-30:
I use an Odrive v3.6 24v with current firmware 4.12 and the motor D5065- 270Kv / Encoder ANT102. There is a gear box on the motor. Therefore, the motor cannot rotate freely but is loaded by the friction torque. 
The encoder index search and encoder calibration fails (ERROR_ENCODER_FAILED/ ERROR_INDEX_NOT_FOUND_YET). I tried to increase the current during calibration using the parameters current_lim and calibration_current but this did not solve the problem. My power supply (24v @ 10 A) shows about 300-400 mA during calibration, even though the parameters current_lim was set to 10 A and calibration_current to 5 A.
It seems that these parameters have no effect in the current software version.

tombsar, 2020-09-30:
The latest firmware is actually v0.5.1 (https://github.com/madcowswe/ODrive/releases/tag/fw-v0.5.1) that just released. Perhaps it's been fixed in that?

sl13, 2020-10-01:
Many thanks for the hint. 
I have now tried the latest version of the firmware but it seems to me that there are no changes regarding the parameter calibration_current. 
But I must also correct my error description.
The current changes by changing the parameter but never reach the specified value. 
I was a bit careful in the beginning and only experimented with very small values. 
Therefore, I did not recognize the change of the current.
Now I tried to increase the value significantly and now I got higher currents at the motor as well. 
Following a series of measurements on my setup:
calibration_current		measured 
10 A				0,25 A
12,5 A				0,42 A
15 A				0,64 A
17,5 A				0,91 A
20 A				1,23 A
30 A				3,08 A
It’s strange that the measured value is so far away from the setpoint of calibration_current. I have also tried to recalibrate the motor because I think it is a problem with the phase resistance estimation. I got the following values:
Resistance = 0.04075 Ohm
Inductance = 1.478 *10^-5 H
Are these realistic values for a motor of type D5065- 270KV? What else could be the cause of the problem?

samuelsadok, 2020-10-02:
This is expected.

Think of it like this: if you don't put mechanical load on the motor then the output power of the motor is zero (by definition of mechanical power). Therefore the electrical power coming from the DC supply must also be zero, but because the DC supply has non-zero voltage, its current must be zero. Despite this you can still have non-zero current in the motor, it just circulates in the motor coils forever.

In reality of course you have losses everywhere, especially the phase resistance. That's why you see DC current go up slightly if you increase the motor current: you just dump more power into heating the motor coils.

Also see the side note in [this section](https://docs.odriverobotics.com/#configure-m0).

The calibration values you posted are good.

sl13, 2020-10-02:
This would be expectable if you use the motor in combination with a close loop control.
In open loop control it should be a different story.  I think in calibration mode (index_search or index_offset_calibration) odrive uses open-loop control, so that the controller does not use any feedback from the encoder or the back emf of the motor windings. The oDrive unit should then behave like a stepper motor controller. Depending on the desired motor angle the voltage and based on this the current of the three phases should change independently from the mechanical load on the motor.
Due to the very slow speed during calibration the effect of voltage induction and with that the reduction of the current through the windings shouldn’t be significant.

samuelsadok, 2020-10-02:
The physical constraint "DC-power == mechanical power + losses" always applies, no matter how the controller is implemented, as long as there's no energy storage on the controller itself (there actually is, but it's super small). Otherwise the conservation of energy would be violated.

The difference between closed loop control and open loop control is that in closed loop mode the current vector is controlled to lead the rotor flux vector by 90° to achieve max torque-per-amps, whereas in open loop mode the rotor flux vector "snaps" to the same angle as the current vector (assuming no mechanical load).

The term "open loop control" is a bit misleading here: Only the current command is generated in open loop mode, but the current controller itself still operates in closed loop mode, using the two phase current sensors on the board as two-dimensional feedback.

So basically we're doing the green circled part here:
[image]
([original diagram](https://docs.odriverobotics.com/control))

sl13, 2020-10-05:
> „Only the current command is generated in open loop mode, but the current controller itself still operates in closed loop mode, using the two phase current sensors on the board as two-dimensional feedback.“

Ok, so the current control loop is active. This is exactly what I mean. We have a current setpoint for the current control loop defined by the parameter calibration_current and the current controller tries to apply this current to the motor windings. It doesn’t matter if the motor produces any mechanically power or not (the motor doesn't even have to turn). The defined current should flow. When there is no load on the motor, the current through the windings should produce only heat “power losses”. This is also the point mentioned in the documentation: “to the largest value you feel comfortable leaving running through the motor continuously when the motor is stationary.” [doc](https://docs.odriverobotics.com/#configure-m0). I also looked in the implementation of the encoder offset calibration, there you can find the equitation which calculate the necessary DC voltage depending on the phase resistance of the motor to drive the current defined by the parameter calibration_current [link](https://github.com/madcowswe/ODrive/blob/7831d795235e5ef8535e4b46621a0721b458ec8f/Firmware/MotorControl/encoder.cpp#L197). I also tried out what happens when I apply mechanical load to the axis of the motor. The current doesn’t increase. I could only observe that the motor skips at an overload situation one pol pair, like a stepper motor losing steps by mechanically overload. When I now collected all this information, I came back to my stating point and believe that there is something wrong with the implementation of the parameter calibration_current.

samuelsadok, 2020-10-05:
> [...] there you can find the equitation which calculate the necessary DC voltage depending on the phase resistance of the motor to drive the current defined by the parameter calibration_current link.

This algorithm calculates not the DC voltage but the AC voltage (the amplitude of the voltage from the motor phases to the motor's internal star point).

This actually reminds me that we're not doing closed loop current control in encoder calibration yet. But the open loop motor current should be still in the right ballpark, provided that phase resistance is configured correctly.

> The defined current should flow.

"... in the motor". Correct. If you take a current clamp and put it around one of the motor phases you should see a sinusoid with an amplitude of `calibration_current` Amps.

However you mentioned that you were looking at the current indicator of your _DC power supply_. That is not the same. The DC (power supply) current can be almost zero while the AC (motor) current is 10A or more. This was what I was trying to point out with the input-vs-output power reasoning.

If you don't have a current clamp you can try this: Hold the motor and do calibration with 10A. Then change calibration_current to 30A and do it again. The motor should feel more aggressive now, exerting 3x the torque compared to before.

sl13, 2020-11-04:
> That is not the same. The DC (power supply) current can be almost zero while the AC (motor) current is 10A or more. This was what I was trying to point out with the input-vs-output power reasoning.

This week I got some time to check this and it look like you are right. 
I don't have a current clamp, so I used three resistors with 235 mOhm and an oscilloscope to measure the voltage over the resistors. Like I post before, my phase resistance is 0.04075 Ohm without the shunt. After adding the shunts, the calibration routine measures 0.41602 Ohm. The value is a little bit higher as I would expect it but it is in the correct range. The following two pictures shown the voltage drop over one shunt during the index search. 

[image]
[image]

In this image you can see a peak voltage of around 3 to 4 V @ 0.235 Ohm --> 12,8 to 17 A by a value of 8 A for the parameter calibration_current.
The next two pictures shown the voltage drop during the offset calibration routine. 

[image]
[image]

The voltage value is a bit more inconsistence but also around 3 V, so the same 12,8 A like in the previous calibration routine. 
By these results, I could eliminate my doubts, that the value calibration_current has no effect or in a different way than the documentation said. But still, the torque of the motor during calibration feels like it is much lower than in normal close loop operation even with the same value for the current.

closed 2020-11-04 by sl13