odriverobotics/ODrive #506: encoder.set_linear_count sideeffects / race condition ?

dronus, 2020-10-13:
Is encoder.set_linear_count allowed to be used while control is engaged?

-If using encoder.set_linear_count to reset axes origin sometimes, most likely in axis idle state, anything works as expected.

-If using encoder.set_linear_count to repeatedly do small offset calibrations while moving, the drive sometimes stall. It will not react to move_to_pos anymore, despite showing some current draw. I interpret this as offset calibration gone away.

Is this as intended, like set_linear_count should only be used in idle state or somewhat?

dronus, 2020-10-13:
My guess is, that while Encoder::set_linear_count is protected from interrupts, Axis::run_control_loop() and hence finally Encoder::update is not, and may give yield to user command input, calling  Encoder::set_linear_count ?  This may mess up the computation of delta_enc, affecting count_in_cpr_, affecting interpolated_enc, and finally phase_.  But this is just a guess after looking at the code.

dronus, 2020-10-13:
However, this would also not show a suspectability to doing set_linear_count while moving, as Encoder::update is always used. Maybe it's just the amount of calls that at some point will randomly fail by corruption of count_in_cpr_.

samuelsadok, 2020-10-14:
The USB communication thread runs at a lower priority than the axis update thread so if I'm not mistaken `Encoder::update()` cannot be interrupted by a USB-originating call to `Encoder::set_linear_count()`.

However note that `set_linear_count(0)` acts "below" the encoder offset calibration. That means if you have a certain offset calibrated, then turn the axis to a non-zero position (let's say `delta`) and do `set_linear_count(0)`, you would also need to adjust the calibrated offset by `-delta` (possibly subject to unit conversions).

So unless there is drift between _encoder vs motor_ (as opposed to axis vs external reference frame) `set_linear_count` is not suitable for your case. In that case I think your best bet would be to just do the conversion between your machine's reference frame and ODrive's reference frame on the device that commands the ODrive.

dronus, 2020-10-14:
> However note that `set_linear_count(0)` acts "below" the encoder offset calibration. That means if you have a certain offset calibrated, then turn the axis to a non-zero position (let's say `delta`) [...]

Not sure about this. The delta is computed by  line https://github.com/madcowswe/ODrive/blob/b4e5469dcf6e2f2139cd0ae9f20e8bf6bbb256c2/Firmware/MotorControl/encoder.cpp#L449
and both of this counters are updated by `set_linear_count`. So if I get that right, `set_linear_count` will not affect the `count_in_cpr_` value, thus should not change phase offset. That also explains why `set_linear_count` works most of the time without affecting the calibration. But _sometimes_ it is not.

As you said, it would be better to go for an own reference frame implementation. However, this may lead to value overflow / precision issues on the long term, as float's precision is lower when offset is high.

samuelsadok, 2020-10-14:
Hmm yeah you're right, I mixed up `set_linear_count()` and `set_circular_count()`. Well then I don't have an immediate guess as to what might cause this drift. The encoder implementation definitely needs some clean up though, so if it's not a blocking issue I would prefer to get the encoder refactoring done first.

Regarding the float precision concerns, have you considered using [circular position control](https://docs.odriverobotics.com/#circular-position-control)?

Wetmelon, 2021-09-13:
There has been major refactoring of threading since this topic came up.  I'll close it for now until we see an issue with the new threading stack

closed 2021-09-13 by Wetmelon