odriverobotics/ODrive #508: Driver Firmware Update IndexError

jlhamilt, 2020-10-18:
When trying to update my device firmware to match the new installation of the software, the update failed due to an index error and now my driver will not connect. Please advise.

```
>odrivetool dfu
ODrive control utility v0.5.1
Waiting for ODrive...
Downloading json data from ODrive... (this might take a while)
Found ODrive 2070399F4D4D (v3.6-56V) with firmware v0.4.12-dev
Checking online for newest firmware... found v0.5.1
Downloading firmware v0.5.1...
Saving configuration to C:\Users\JHAMI_~1\AppData\Local\Temp\odrive-config-2070399F4D4D.json...
Configuration saved.
Putting device 2070399F4D4D into DFU mode...
Still waiting for the device to reappear.
Use the Zadig utility to set the driver of 'STM32 BOOTLOADER' to libusb-win32.
Erasing... done
Flashing... done
Verifying... (sector 1/6)
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
C:\Users\jhami_000\Anaconda3\Scripts\odrivetool in <module>()
    141         print_version()
    142         import odrive.dfu
--> 143         odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
    144
    145     elif args.command == 'liveplotter':

C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfu.py in launch_dfu(args, logger, cancellation_token)
    455     firmware = FirmwareFromFile(args.file) if args.file else None
    456
--> 457     update_device(device, firmware, logger, cancellation_token)
    458
    459

C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfu.py in update_device(device, firmware, logger, cancellation_token)
    392         for i, (sector, expected_data) in enumerate(touched_sectors):
    393             print("Verifying... (sector {}/{})  \r".format(i, len(touched_sectors)), end='', flush=True)
--> 394             observed_data = dfudev.read_sector(sector)
    395             mismatch_pos = get_first_mismatch_index(observed_data, expected_data)
    396             if not mismatch_pos is None:

C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfuse\DfuDevice.py in read_sector(self, sector)
    192         Returns: a byte array containing the data
    193         """
--> 194         self.set_alternate_safe(sector['alt'])
    195         self.set_address_safe(sector['addr'])
    196

C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfuse\DfuDevice.py in set_alternate_safe(self, alt)
    148     def set_alternate_safe(self, alt):
    149         self.set_alternate(alt)
--> 150         if self.get_state() == DfuState.DFU_ERROR:
    151             self.clear_status()
    152             self.wait_while_state(DfuState.DFU_ERROR)

C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfuse\DfuDevice.py in get_state(self)
     65
     66     def get_state(self):
---> 67         return self.control_msg(DFU_REQUEST_RECEIVE, DFU_GETSTATE, 0, 1)[0]
     68
     69     def abort(self):

IndexError: array index out of range
```

tombsar, 2020-10-19:
Unfortunately, trying to flash an odrive from Windows soft-bricks it right now, as you've discovered. The only fix I found was using a Linux PC to perform the upgrade, which brought it back to life.

See https://github.com/madcowswe/ODrive/issues/179#issuecomment-683354557.

Wetmelon, 2020-10-20:
> 
> 
> Unfortunately, trying to flash an odrive from Windows soft-bricks it right now, as you've discovered. The only fix I found was using a Linux PC to perform the upgrade, which brought it back to life.
> 
> See [#179 (comment)](https://github.com/madcowswe/ODrive/issues/179#issuecomment-683354557).

I'll check this out, but I use windows and I just flashed my ODrive fine... we do check these things before we ship ;)

samuelsadok, 2020-10-20:
@jlhamilt You should be able to recover the ODrive by flicking the DFU switch and then power cycling it and rerunning the DFU command.

The problem you had with the DFU process is something I've seen before, but not often enough to be able to chase it down. In fact I would expect it to go away if you try again. If not, you can try this: edit `C:\Users\jhami_000\Anaconda3\lib\site-packages\odrive\dfuse\DfuDevice.py` line 66-67 by replacing
```
    def get_state(self):
        return self.control_msg(DFU_REQUEST_RECEIVE, DFU_GETSTATE, 0, 1)[0]
```
with
```
    def get_state(self):
        import time
        while True:
            msg = self.control_msg(DFU_REQUEST_RECEIVE, DFU_GETSTATE, 0, 1)
            if len(msg) > 0:
                return msg[0]
            time.sleep(0.1)
```

Then rerun the DFU command.

samuelsadok, 2020-10-21:
I just published a hotfix release of odrivetool (see https://github.com/madcowswe/ODrive/issues/509#issuecomment-713459343). Besides other things it includes a similar workaround to the one I described above. So let me know if you still get DFU issues with that.

Wetmelon, 2020-11-10:
I'm going to close this due to no further issues reported.  Please re-open it if you continue to have problems.

closed 2020-11-10 by Wetmelon