odriverobotics/ODrive #509: v0.5.1 Firmware - Downloading json data from ODrive...

BENJI16, 2020-10-19:
ODrive: 3.5 - 48V
Original Firmware: v0.4.7-dev

I tried running the full calibration and was getting a python error about thermistors, so following the tutorials I went to update the firmware. When trying the odrivetool dfu it recognized the board, but gave me this error:

sudo odrivetool dfu
ODrive control utility v0.5.1
Waiting for ODrive...
Downloading json data from ODrive... (this might take a while)
Found ODrive 206130804648 (v3.5-48V) with firmware v0.4.7-dev
Checking online for newest firmware...'_sre.SRE_Match' object is not subscriptable
'_sre.SRE_Match' object is not subscriptable
'_sre.SRE_Match' object is not subscriptable
'_sre.SRE_Match' object is not subscriptable
Traceback (most recent call last):
  File "/usr/local/bin/odrivetool", line 143, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/usr/local/lib/python3.5/dist-packages/odrive/dfu.py", line 457, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/usr/local/lib/python3.5/dist-packages/odrive/dfu.py", line 316, in update_device
    raise Exception("could not find any firmware release for this board version")
Exception: could not find any firmware release for this board version

So thinking it was an issue with downloading the firmware, I manually downloaded the latest .hex firmware file and it seemed to complete successfully:

`sudo odrivetool dfu ~/Downloads/ODriveFirmware_v3.5-48V.hex 
ODrive control utility v0.5.1
Waiting for ODrive...
Downloading json data from ODrive... (this might take a while)
Found ODrive 206130804648 (v3.5-48V) with firmware v0.4.7-dev

Warning: you are about to flash firmware [unknown version] which is older than the firmware on the device (v0.4.7-dev).
Do you want to flash this firmware anyway? [y/N] y
Saving configuration to /tmp/odrive-config-206130804648.json...
Configuration saved.
Putting device 206130804648 into DFU mode...
Erasing... done            
Flashing... done            
Verifying... done            
Waiting for the device to reappear...
Downloading json data from ODrive... (this might take a while)
Restoring configuration from /tmp/odrive-config-206130804648.json...
Could not restore axis1.config.spin_up_target_vel: property not found on device
Could not restore axis1.config.counts_per_step: property not found on device
Could not restore axis1.config.spin_up_current: property not found on device
Could not restore axis1.config.ramp_up_time: property not found on device
Could not restore axis1.config.spin_up_acceleration: property not found on device
Could not restore axis1.config.ramp_up_distance: property not found on device
Could not restore axis1.trap_traj.config.A_per_css: property not found on device
Could not restore axis1.controller.config.setpoints_in_cpr: property not found on device
Could not restore axis1.encoder.config.idx_search_speed: property not found on device
Could not restore axis0.config.spin_up_target_vel: property not found on device
Could not restore axis0.config.counts_per_step: property not found on device
Could not restore axis0.config.spin_up_current: property not found on device
Could not restore axis0.config.ramp_up_time: property not found on device
Could not restore axis0.config.spin_up_acceleration: property not found on device
Could not restore axis0.config.ramp_up_distance: property not found on device
Could not restore axis0.trap_traj.config.A_per_css: property not found on device
Could not restore axis0.controller.config.setpoints_in_cpr: property not found on device
Could not restore axis0.encoder.config.idx_search_speed: property not found on device
Some of the configuration could not be restored.
Configuration restored.
Device firmware update successful.
`

Before powering it off I checked and the firmware was the latest v0.5.1. However after power cycling the odrivetool gets hung on:

odrivetool
ODrive control utility v0.5.1
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

Downloading json data from ODrive... (this might take a while)

It never gets past this line, so it does recognize that an ODrive was connected, but I can't check anything or reflash the firmware. Is there an easy way to resolve this or is the ODrive out of commission?

Wetmelon, 2020-10-20:
I think something happened after you flashed 0.5.1, AFTER it connected the first time and tried to restore your configuration.  Can you maybe try to flash 0.4.12 and then do an `odrv0.erase_configuration()`, then flash `0.5.1`?

bblacklock3, 2020-10-20:
It can't find the odrive at all, so I'm unable to flash anything right now. I just ordered an STLink, so hopefully that will work.

BENJI16, 2020-10-20:
@Wetmelon Have there been any fixes for the first error I encountered after running odrivetool dfu? I have another board stuck on v0.4.7, so unless there's a way to revert the odrivetool back to 4.7 it looks like I have to update it and risk soft bricking another board.

samuelsadok, 2020-10-20:
@BENJI16 @bblacklock3 You should be able to recover your ODrive even without an STLink by flicking the DFU switch on the board. To be sure, disconnect all GPIOs from the ODrive (especially GPIO6 can mess with DFU). Then run odrivetool's DFU as usual. As @Wetmelon suggested, flashing a known-good version and erasing the config would be a good idea.

Once you start configuring the ODrive again and it fails to reboot again, it would be helpful to have a list of configs that you changed so we can try to reproduce it.

@BENJI16  I haven't seen this firmware download error before. I tried on my machine with the same Python version (3.5) and it works fine. To help debugging you can edit `/usr/local/lib/python3.5/dist-packages/odrive/dfu.py`, line 193 by replacing `print(ex)` with `raise`. You can also add `print(hw_version_match)` after line 151. Then run the tool again and let me know the output.

PS: I edited your comment to use code-block formatting (triple backticks), makes it easier to understand ;)

Wetmelon, 2020-10-20:
@BENJI16 to confirm, you've already upgraded odrivetool (`pip install --upgrade odrive`) right?

BENJI16, 2020-10-20:
So I've been trying mostly with the 0.5.1 tool, but I did also try it with 0.4.12. After flipping the switch into DFU mode and running odrivetool dfu with 0.5.1 this has always happened since the first time I tried it:

```
odrivetool dfu
ODrive control utility v0.5.1
Waiting for ODrive...
Traceback (most recent call last):
  File "/home/ben/.local/bin/odrivetool", line 143, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 451, in launch_dfu
    find_odrive_cancellation_token, cancellation_token)
  File "/home/ben/.local/lib/python3.5/site-packages/fibre/discovery.py", line 179, in find_any
    return result[0]
IndexError: list index out of range
```

I didn't see this error posted around anywhere so I just assumed it wasn't able to find the board in DFU mode. Just now I tried it with 0.4.12 and got a different error that I think was previously an issue:

```
odrivetool dfu
ODrive control utility v0.4.12
Waiting for ODrive...
Traceback (most recent call last):
  File "/home/ben/.local/bin/odrivetool", line 142, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 454, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 258, in update_device
    serial_number = device.serial_number
  File "/usr/local/lib/python3.5/dist-packages/usb/core.py", line 852, in serial_number
    self._serial_number = util.get_string(self, self.iSerialNumber)
  File "/usr/local/lib/python3.5/dist-packages/usb/util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)
```

BENJI16, 2020-10-20:
I added print(hw_version_match) above the line it's getting stuck on for 0.5.1:

```
odrivetool dfu
ODrive control utility v0.5.1
Waiting for ODrive...
Downloading json data from ODrive... (this might take a while)
Found ODrive 207D37A63548 (v3.5-48V) with firmware v0.4.7-dev
Checking online for newest firmware...<_sre.SRE_Match object; span=(0, 27), match='ODriveFirmware_v3.5-24V.hex'>
Traceback (most recent call last):
  File "/home/ben/.local/bin/odrivetool", line 143, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 460, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 317, in update_device
    firmware = get_newest_firmware(hw_version)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 198, in get_newest_firmware
    firmwares = list(firmwares)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 186, in get_all_github_firm
wares
    fw = FirmwareFromGithub(release_json, asset_json)
  File "/home/ben/.local/lib/python3.5/site-packages/odrive/dfu.py", line 148, in __init__
    self.hw_version = (int(hw_version_match[1]),
TypeError: '_sre.SRE_Match' object is not subscriptable
```
hw_version_match: _sre.SRE_Match object; span=(0, 27), match='ODriveFirmware_v3.5-24V.hex
It looks like it's complaining since I have 48v board and it sees the 24v fw, but I'm not sure why it's not just finding the 48v fw...

BENJI16, 2020-10-20:
> I think something happened after you flashed 0.5.1, AFTER it connected the first time and tried to restore your configuration. Can you maybe try to flash 0.4.12 and then do an `odrv0.erase_configuration()`, then flash `0.5.1`?
> 
> Worst case, you'll need an STLink v2 to flash via the programming header.

After adding the recent dfu fixes (langid udev issue, index out of range) it was able to flash 0.5.1 fw while starting in dfu mode, so it's no longer stuck! However after successfully repeating what you suggested: 0.4.12, erase_config, then 0.5.1 it still gets stuck when running odrivetool:

```
odrivetool
ODrive control utility v0.5.1
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().
Downloading json data from ODrive... (this might take a while)
```
So at this point it must be something to do with 0.5.1 fw right?

samuelsadok, 2020-10-20:
Thanks for the stack traces! I remember I actually _did_ see two of these issues before (`'_sre.SRE_Match' object is not subscriptable` and `IndexError: list index out of range`) and they are both fixed by 96cbe69f8aaabba669e39331e15ca364dd4e6eba. I'll make sure we release a hotfix on pip which includes these fixes.

> After adding the recent dfu fixes [...]

I assume you're referring to the commit I linked above?

> So at this point it must be something to do with 0.5.1 fw right?

So it seems. At this point the most efficient path forward would be to attach an STLink/v2 debugger and then you could just pause the execution and see where it gets stuck (HardFault_Handler is a usual suspect) complete with stack trace. If you don't have an STLink I can just give some general pointers:

 -  Do you have any SPI encoder connected? There used to be issues with that but as far as I know they were fixed before the 0.5.1 release.
 - Run `odrivetool` with the `--verbose` flag.
 - If you have access to another operating system or another computer, try on there.

BENJI16, 2020-10-20:
> I assume you're referring to the commit I linked above?
Yup, those fixes and adding the udev rule you posted somewhere else did the trick!

I ordered an STLink yesterday, so that might come in at somepoint, but this is interesting:
```
odrivetool --verbose
ODrive control utility v0.5.1
Waiting for ODrive...
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().
ConfigurationValue 1
        InterfaceNumber 0,0
                EndpointAddress 130
        InterfaceNumber 1,0
                EndpointAddress 1
                EndpointAddress 129
        InterfaceNumber 2,0
                EndpointAddress 3
Unexpected exception after discovering channel: Traceback (most recent call last):
  File "/home/ben/.local/lib/python3.5/site-packages/fibre/discovery.py", line 113, in did_discover_channel
    with open(cache_path, 'w+') as json_cache:
PermissionError: [Errno 13] Permission denied: '/home/ben/.cache/odrivetool/fibre_schema_cache_2604684226'
```
Is this permission error something you guys have seen before?

samuelsadok, 2020-10-20:
Hmm this is interesting indeed. Can you check `ls -la /home/ben/.cache/odrivetool/fibre_schema_cache_2604684226`? I could imagine that it's owned by root if you previously ran odrivetool with `sudo`.

You should be able to fix the permissions with
```
sudo chown $USER:$USER ~/.cache/odrivetool/fibre_schema_cache_2604684226
chmod u+rw ~/.cache/odrivetool/fibre_schema_cache_2604684226
```

BENJI16, 2020-10-20:
Success finally!

I guess when I tried sudo odrivetool before that changed the permissions? I think that was the only issue here besides the dfu fixes (which likely caused me try sudo odrivetool) since I redid the fw updates making sure to reboot in between and it gave the same result.

Firmware is up to date and I don't see any errors now!

samuelsadok, 2020-10-20:
Sweet! Sorry you were facing such a slew of bugs.

Arguably odrivetool should print a warning instead of silently failing to connect in case of permission errors like this. I added a commit on the hotfix branch to address this (4d59959e07456f7aecefa3a511477f7473943e2f).

samuelsadok, 2020-10-21:
The hotfix is now released as `0.5.1.post0` on PyPI (install with `sudo pip3 install --upgrade odrive`).

It includes these commits:
 - 01174d3597d33d02838f4bcd9370764425e83eed
 - 4d59959e07456f7aecefa3a511477f7473943e2f
 - 40f9fbb6f58588196cabaad0616ac0b8ce4c05e7
 - 2d2892f253c00da9dbfcf518450f6f0a7108ce3f

closed 2020-10-21 by samuelsadok