odriverobotics/ODrive #512: The controller doesn't work

gpistotnik, 2020-10-28:
I think I received faulty controller.

The motor doesn't turn after setup, on osciloscope, I don't even get output for motor. What can be issue?

Wetmelon, 2020-10-29:
Hi @gpistotnik, can you give us a little more information?

* What model of ODrive do you have?
* What motor are you trying to control?
* How are you communicating to the ODrive?
* If you're using odrivetool, can you try `dump_errors(odrv0)` to check for errors?

Thanks!

gpistotnik, 2020-10-29:
Odrive 3.6, communication via USB, I tried to control several hobby BLDC outrunners and https://www.ebay.com/itm/1PC-Nema17-BLDC-Motor-42BLF01-4000RPM-1-9A-24VDC-26W-47mm-Φ5mm/264438939575?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649.

When I use dump_errors(odrv0), I get error after motor calibration (which does not even move the motor) that phase induction is 0 or something like that.

Wetmelon, 2020-10-30:
It looks like these are very low current (high resistance) motors.  Please see the error codes and the instructions described here https://docs.odriverobotics.com/api/odrive.motor.error to try to get them to calibrate properly.

Please be aware that with default settings, ODrive will see +/- up to 0.5A of noise on the current shunts, swamping the measurements for small motors like these.

Wetmelon, 2020-11-10:
I'm going to close this, since I haven't heard any further issues.  Please reopen it if you continue to have problems.

gpistotnik, 2020-11-12:
So I managed to get the motors to work, but they only work on M0, I can swap them all I want, copy configuration from one to another, ... and still, only M0 works. What am I doing wrong?

Wetmelon, 2020-11-13:
What does "work" and "not work" mean?  Are you getting errors during calibration?  If so, which ones?

gpistotnik, 2020-11-13:
On M0 everything works. I mean, i can move with closed loop control and set speed. On M1 I can't even get motor to turn a little in calibration. I have changed motors, recalibrated M0 to another motor, everything fine, but only on M0, M1 does not work here. Motor doesn't turn even for 1 degree

Wetmelon, 2020-11-14:
Can you please use the `dump_errors(odrv0)` command to check for errors on M1?

gpistotnik, 2020-11-18:
Everything looks fine in dump_errors

closed 2021-05-22 by PAJohnson