odriverobotics/ODrive #518: Reproducing the 0.5.1 release build

combinatorylogic, 2020-11-06:
Hi!

I would like to find out the exact steps to reproduce the binary for 0.5.1 release from the git repo (tag fw-v0.5.1, right?). My setup is: ubuntu18.04, with all the packages installed as per developer-guide.md:

`gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (GNU Tools for Arm Embedded Processors 7-2018-q3-update)`

The binary I'm getting is significantly different from the `ODriveFirmware_v3.6-56V.elf` that odrivetool downloads (comparing them with `arm-none-eabi-nm`).

The reason I'm digging into this is that the official binary works fine with an AMT23 absolute encoder attached, while the one we build locally, with all the same configuration and physical setup, yields very high spi error rate. Trying to find out what's the difference here.

Thanks!

Wetmelon, 2020-11-10:
It looks like that's the release I built manually because we didn't have the CI setup properly at the time.  I believe I used arm-none-eabi-gcc 9.2.1

combinatorylogic, 2020-11-10:
Ok, thanks! It explains some of the differences in binaries, but not the SPI difference. I managed to get rid of the SPI errors by changing the SPI clock divider to 64. Does it mean there's some difference in the .ioc file (clock tree settings?) and code generated from .ioc in the official binary build?

Wetmelon, 2020-11-12:
If you re-generate with the .ioc, it will overwrite a bunch of custom code and I would be surprised if you could even compile.  If you do compile, it won't run.  Do not recommend :)

I think the most recent changes will be completely de-coupling from CubeMX anyway.

closed 2020-11-12 by Wetmelon

combinatorylogic, 2020-11-12:
Ok, then it's very strange - the difference is still unexplainable, as default divider in 0.5.1 is 32...