odriverobotics/ODrive #522: Enable current control checks during motor calibration

samuelsadok, 2020-11-12:
Currently we don't run current limit checks during motor calibration. If the motor phases are accidentally shorted this could lead to oscillation and fry the ODrive.

With the integral gain `kI = 10.0f` that's currently used, the current overshoots if (and only if) the motor snaps to align with I_alpha.

[image]

madcowswe, 2020-11-13:
We should probably reduce the `kI` a bit.

samuelsadok, 2020-11-16:
Ok I changed kI to 1.0 in 18a391c092a2dfcd190fe25077f8b3c119427ed2.

For comparison, here's the log with kI = 2.0 and kI = 1.0:

[image]
Maximum observed I_alpha: 13.5 A

[image]
Maximum observed I_alpha: 11.3 A

The total duration of the motor resistance calibration is 3s (24'000 samples) so there is still easily enough time for the controller to settle.

samuelsadok, 2020-11-16:
This is merged to `devel` now.

closed 2020-11-16 by samuelsadok