odriverobotics/ODrive #525: Firmware v0.5.1 causes index pin to be pulled down

kylebme, 2020-11-14:
Firmware version 0.5.1 causes the index pins of the encoder to not reach 3.3V when the encoder sends a pulse. The index pin is limited to around 1.2V (verified with oscilloscope), which is not high enough for the ODrive to see the index pulse.

Switching to FW 0.4.12 with no other changes allows the index pulse to reach 3.3V.

ODrive V3.6 24V
OS: Mac Catalina
ODrive tool from 0.5.1 release

The encoder is the iC-Haus TW28 interpolator chip (with a Renishaw optical encoder) with a custom breakout board. I did a lot of troubleshooting to make sure the issue was not due to my breakout board or schematic. Since the issue gets fixed when changing the firmware to 0.4.12, it's clear that the root cause of the issue is in firmware.

madcowswe, 2020-11-18:
Were you using the M0 or M1 encoder port, or does it happen on both?

kylebme, 2020-11-19:
I tested it mostly on M0 but I tested it once on M1 and it appeared to do the same thing.

samuelsadok, 2020-12-02:
What happens if you disconnect the index pin and measure its voltage on the ODrive side? When I try it I see 3.3V so in this case the ODrive does not pull down the pin. If you end up seeing 0V instead, can you also try with a cleared configuration?

Wetmelon, 2021-07-14:
Any updates to this for 0.5.2?

samuelsadok, 2021-09-03:
Closing this because of inactivity and insufficient information to reproduce. If the issue still persists feel free to comment with new information, then I'll reopen it.

closed 2021-09-03 by samuelsadok