odriverobotics/ODrive #527: Firmware on STM32 Flash is modified unexpectly

EMHacker-alt, 2020-11-16:
After using ```make flash``` command flash the firmware binary of ODrive to the STM32, the STM32 have probability that cannot boot up normally. After reflash the binary again, the operation of STM32 can be recovery. I try to read the FLASH memory of abnormal board using STM32CubeProgrammer and found that a lot of memory was modified.
[image]

Until now, I cannot find any clue that what kind of situation can causing this problem. I usually find this problem after several days using of ODrive. The ODrive board are custom single axis board. And I had produced 30 pcs and found 6 of them have this problem. I also found that if a board have this problem and reflash the binary to a board, the board can have this problem again.

**Desktop (please complete the following information):**
 - OS: MacOS Catalina
 - latested odrivetool Version

samuelsadok, 2020-11-18:
I haven't seen this on original hardware before so it might be caused by an unstable STM32 voltage supply.

I think you can add `-c 'flash verify_image $(FIRMWARE)' \` after the line `-c 'flash write_image erase $(FIRMWARE)' \` in Firmware/Makefile and this should warn you if the written image is not correct. FYI odrivetool is not involved when you type `make flash`.

Wetmelon, 2020-11-19:
Although, it is quite common for ODrive boards that have been sitting without power for some time to not boot up correctly until after a re-flash.

EMHacker-alt, 2020-11-19:
Sorry! My previous description may not clear.  I mean that after download the firmware to the board successfully, and the board works normally, the FLASH memory of STM32 have probability become abnormal. This situation may be days between normally to abnormal.

Jconn, 2021-03-29:
This issue is hitting us as well. @Wetmelon do you know root cause of why a powered off odrive would require a reflash?

Wetmelon, 2021-03-29:
> This issue is hitting us as well. @Wetmelon do you know root cause of why a powered off odrive would require a reflash?

It's unknown. Some people just have bad luck with it. I suspect it may have something to do with inrush current or something funky like that. Have you tried a précharge resistor?

927tang, 2022-04-29:
We are struggling with this issue too. @Wetmelon  and we have done some experiments and use a relay to switch off/on the power every 10s. This issue comes up about once a day on some boards (V3.6-56V fw0.5.1) repeatly. Has anyone hitted by the issue and found the root cause.

closed 2022-09-21 by Wetmelon