odriverobotics/ODrive #535: Fibre-over-CAN

madcowswe, 2020-11-26:
CAN is more robust than USB and as such more suitable as the primary interface for ODrive. To make the full feature set of ODrive available over CAN, we want to run Fibre over CAN, possibly ISO-TP.

moiflo033, 2021-03-23:
May i suggest using UAVCAN? 
It is a protocol that supports multiple and redundant transports (for example CAN2.0B, CAN FD, Ethernet) based on a publish/subscribe architecture.
Sry if i missed the point of this issue.
Here are some links you may find useful:
[Uavcan.org](https://uavcan.org/)
[libcanard (C implementation of UAVCAN)](https://github.com/UAVCAN/libcanard)

borongyuan, 2021-07-13:
I agree that CAN is more robust than USB. However, many IPCs do not have native CAN. So we need to add a PCIe-CAN module or USB-CAN module. If we use a USB-CAN module, USB is still used. It cannot be more reliable than using USB directly. But it should be more easier to use if CANOpen 402 is supported.
For robot development, I think a good way is to add support for ros_control or ros2_control (which I'm currently working on). Perhaps a better choice is to run ROS directly on ODrive (aka [micro-ROS](https://micro.ros.org)). In that case, the communication is handled by DDS, which can be very reliable.

Wetmelon, 2021-07-14:
@moiflo033 ODrive already uses Fibre on USB and UART, this is just extending that functionality to CAN.  This does not preclude the use of other CAN protocols over the interface (it was designed to be easily switchable from the outset).  I looked into UAVCAN a while ago and it didn't really make any sense with ODrive and would be hard to integrate, but I can check again.

@borongyuan We have some exciting developments coming for you :)

borongyuan, 2021-08-01:
@Wetmelon Thanks for your reply.
This is our initial release.
[odrive_ros2_control](https://github.com/Factor-Robotics/odrive_ros2_control)
We are glad to add CAN support when new protocols are ready.

closed 2021-09-13 by Wetmelon