odriverobotics/ODrive #539: SPI Encoders randomly stop working after reset (nRST)

OstapFerensovych, 2020-12-09:
I am using AS5048A encoders on both axes.
I get ERROR_ABS_SPI_COM_FAIL randomly on one or both axis after I reset the STM chip by nRST pin.
`odrv0.axis0.encoder.error = 0` `odrv0.axis1.encoder.error = 0` do not clear the error. While the error is present spi generates clocks and data is present on MISO from both encoders.
After the power cycle, the errors are gone.

To reproduce this behavior connect two AS5048A encoders. (In my case I use GPIO4 and GPIO8 as nCS pins)
Reset STM32F405 by touching nRST to GND.

 - Firmware v0.5.1-release
 - ODrive HW v3.6-56V
 - OS: Ubuntu 20.10
 - odrivetool Version v0.5.1.post0

OstapFerensovych, 2020-12-09:
After some debugging, I found that AMS encoders have an error flag at the 14th bit, and ODrive firmware checks this flag, but it does nothing when this flag is `1`:
`if (ams_parity(rawVal) || ((rawVal >> 14) & 1)) { return; }`

It looks like while I do reset on the nRST pin - the SPI clock interrupts and AS5048A rises error flag.

Information from AS5048A datasheet.
> Possible conditions which force the ERROR FLAG to be set:
...
-Wrong number of clocks (no full transmission cycle or too
many clocks)
...

So is there a need to check this flag if there is a parity check implemented in firmware?

samuelsadok, 2020-12-09:
Thanks for debugging this!

Looking at the datasheet it's unclear to me what exactly happens when the encoder detects a parity error in a command coming from the master. Is the next response from the encoder a valid angle or is it all zeros? If it's a valid angle then we can indeed just ignore the error flag. If not then we need to ignore the response and somehow send a clear-fault command, otherwise we cannot detect the next time the encoder detects a parity error.

Since I don't have this encoder at hand would you mind testing this? You can flip a bit in `abs_spi_dma_tx_` in encoder.hpp and see if the encoder still returns valid angles.

OstapFerensovych, 2020-12-09:
Yes, I already tested this!
I simply removed ` || ((rawVal >> 14) & 1)` and looks like everything works fine. The angle is valid when the error flag is 1.

I think AMS did this error flag in their chips to detect if there was a successful transmission from MCU to AS5048A by MOSI line. For example, you need to configure something in AS5048A and MOSI line is noisy and unstable, so it also checks for parity, a number of clocks and if something is wrong it rises this flag until MCU reads it and clears.

samuelsadok, 2020-12-10:
Yeah so if the error condition is removed (i.e. we send valid commands again) then the encoder sends valid angle data and just leaves the error flag latched high. So in your case it's fine to remove the check because you know when and why the encoder detected an error.

But I'm not yet sure if this fix is safe in the more general case. Theoretically it's possible that the encoder sends all zeros on the transmission that immediately follows a bad command. In your case this would be the very first (or second?) encoder reading that you receive after ODrive reset so it would go unnoticed. But in general, if the MOSI line really _is_ noisy this could mess with the ODrive's position estimate.

To test this you would need to flip a single bit in `abs_spi_dma_tx_` so that you deliberately trip the encoder's parity check continuously.

Wetmelon, 2020-12-13:
I have indeed seen cases where the encoder appears to send garbage data with the error flag.  We could look into sending a "reset error" command when clearing odrive errors

hbuhle2s, 2021-04-18:
Hi,
I ran into this issue as well and fixed it also by removing the  `|| ((rawVal >> 14) & 1)` check. Would you like me to make a PR that adds an option to disable that check? I think that would help a lot of people and I think it would be nice if that fix can make it into 0.5.2.

samuelsadok, 2021-04-30:
It depends on whether the encoder responds with valid position data if the sender sends an invalid command. If not, then disabling the error flag check is not safe and instead a reset command needs to be sent to the encoder. So someone needs to check this:

> To test this you would need to flip a single bit in abs_spi_dma_tx_ so that you deliberately trip the encoder's parity check continuously.

Wetmelon, 2021-09-13:
We're defaulting to the safe state here.  If a well-tested PR comes along, we will consider making this change.

closed 2021-09-13 by Wetmelon