odriverobotics/ODrive #545: devel branch fails to build on Mac (both intel and arm)

tmagik, 2021-01-11:
**Describe the bug**
Lots of errors like:
``` *** tup messages ***
 *** Command ID=5145 failed with return value 1
*  89) arm-none-eabi-g++ -std=c++17 -Wno-register -c ./communication/communication.cpp
-D__weak="__attribute__((weak))" -D__packed="__attribute__((__packed__))" -DUSE_HAL_DRIVER
-mthumb -mfloat-abi=hard -Wno-psabi -Wall -Wdouble-promotion -Wfloat-conversion -fdata-sections
-ffunction-sections -g -DUSB_PROTOCOL_NATIVE -DUART_PROTOCOL_ASCII -O2
-IThirdParty/FreeRTOS/Source/include -IThirdParty/FreeRTOS/Source/CMSIS_RTOS
-IThirdParty/CMSIS/Include -IThirdParty/CMSIS/Device/ST/STM32F7xx/Include -IThirdParty/CMSIS/Device/ST/STM32F4xx/Include -IThirdParty/STM32_USB_Device_Library/Core/Inc
-IThirdParty/STM32_USB_Device_Library/Class/CDC/Inc -DHW_VERSION_MINOR=5
-DHW_VERSION_VOLTAGE=24 -IBoard/v3/Inc -IBoard/v3/../../ThirdParty/FreeRTOS/Source/portable/GCC/ARM_CM4F
-DSTM32F405xx -DHW_VERSION_MAJOR=3 -IThirdParty/STM32F4xx_HAL_Driver/Inc -DARM_MATH_CM4
-mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -DFPU_FPV4 -I./. -I./MotorControl -I./fibre/cpp/include -o build/._communication_communication.cpp.o
Assembler messages:
Fatal error: can't create build/._communication_communication.cpp.o: No such file or directory
```

**To Reproduce**
On a mac
```
git clone https://github.com/odriverobotics/odrive
cd odrive/Firmware
cp tup.config.default tup.config
echo 'CONFIG_BOARD_VERSION=v3.5-24V' >> tup.config   
make
```
**Expected behavior**
make completes without errors, as it does on -master branch

**Desktop (please complete the following information):**
 - OS: Macbook Air M1 (arm), Big sur 11.0.1, Macbook Pro, (intel)

**Additional context**
tup --version
tup v0.7.10-9-g47d378cf
osxfuse (macfuse) 4.0.5

tmagik, 2021-01-11:
Something in https://github.com/odriverobotics/ODrive/commit/75577940f8437de8cf6d8f2162c371d8e9af9c89 severely breaks the mac build(s)

samuelsadok, 2021-01-11:
Apparently something on macOS was unhappy with the output files starting with a dot. Oddly it passed our CI checks but I could reproduce the issue locally. Thanks for reporting! I pushed a fix here: 33ae4e8c925278bad63f19588707f302e8f3fc87

closed 2021-01-11 by samuelsadok