odriverobotics/ODrive #547: Homing not returning to zero position

fllmn, 2021-01-13:
**Describe the bug**
When running the homing procedure the motor moves until end-stop is pressed and then stops. As the end-stop is still pressed the motor will be unable to go into CLC as the end-stop will generate an error.

The underlying issue seems to be that `input_pos_` and `pos_setpoint_` will get updated to the same value when `start_closed_loop_control` gets called. Since the trajectory generator uses the diff between `input_pos_`and `pos_setpoint_` the planned time is 0 and in turn the motor never moves.

 - OS: Linux 5.9.11-3-MANJARO
 - odrivetool Version: v0.5.1.dev

This could of course only be a problem with the dev branch.

riewert, 2021-01-26:
I have the same problem on devel. I agree with everything you say. I think it may be that the `controller_.pos_estimate_linear_src_` is not being updated correctly, before this line gets called.

https://github.com/odriverobotics/ODrive/blob/devel/Firmware/MotorControl/axis.cpp#L300

Wetmelon, 2021-01-26:
What did you set for `min_endstop.config.offset`?

riewert, 2021-01-27:
I've tried both positive and negative offsets. This is what I am currently running.

```config:
  debounce_ms: 50 (uint32)
  enabled: True (bool)
  gpio_num: 5 (uint16)
  is_active_high: False (bool)
  offset: -1.0 (float)
endstop_state: False (bool)```

I've run through it a few times in the debugger. It looks like one call is made [here](https://github.com/odriverobotics/ODrive/blob/devel/Firmware/MotorControl/controller.cpp#L130) between the [closed loop control starting](https://github.com/odriverobotics/ODrive/blob/devel/Firmware/MotorControl/axis.cpp#L415) and the [trap_traj being set](https://github.com/odriverobotics/ODrive/blob/devel/Firmware/MotorControl/controller.cpp#L180).

riewert, 2021-01-27:
This fixes it on my system.

Also added a small change, the offset position is set to 0 instead of the trigger point.

https://github.com/odriverobotics/ODrive/pull/550

fllmn, 2021-01-27:
Just like @riewert , the endstop configuration was doing anything. I also tried negative and positive offsets. 

Probably should have added my "solution" to the problem, https://github.com/fllmn/ODrive/commit/d7843da1a410a2966d0cf86ed83c24a9173f4182

I would be surprised if this did't break other features but for the time being solved my problem.

samuelsadok, 2021-02-05:
@riewert's fix is now merged so I'm gonna close this. If you find that the issue still persists, please reopen.

closed 2021-02-05 by samuelsadok