odriverobotics/ODrive #548: autocompletion breaks odrivetool

mngr0, 2021-01-18:
I have Odrivetool just installed on a
`Linux debian 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux`

if I try to use autocompletion it prints a few \n and nothing more. after that if I press enter I get an error that ends in 
`NameError: name 'sys' is not defined`
this does not happen if I do not press tab twice.

Thinking that it was an issue on my installation I clone the repo, and installed, same result.

mngr0, 2021-01-18:
here is the full stacktrace

marco@debian:~/ODrive/tools$ odrivetool
ODrive control utility v0.5.1.post0
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

Connected to ODrive 335B314E3536 as odrv0
In [1]: odvr0fdsf
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/usr/local/lib/python3.7/dist-packages/IPython/terminal/ptutils.py in get_completions(self, document, complete_event)
    112             try:
--> 113                 yield from self._get_completions(body, offset, cursor_position, self.ipy_completer)
    114             except Exception as e:

/usr/local/lib/python3.7/dist-packages/IPython/terminal/ptutils.py in _get_completions(body, offset, cursor_position, ipyc)
    128             body, ipyc.completions(body, offset))
--> 129         for c in completions:
    130             if not c.text:

/usr/local/lib/python3.7/dist-packages/IPython/core/completer.py in _deduplicate_completions(text, completions)
    437     """
--> 438     completions = list(completions)
    439     if not completions:

/usr/local/lib/python3.7/dist-packages/IPython/core/completer.py in completions(self, text, offset)
   1817         try:
-> 1818             for c in self._completions(text, offset, _timeout=self.jedi_compute_type_timeout/1000):
   1819                 if c and (c in seen):

/usr/local/lib/python3.7/dist-packages/IPython/core/completer.py in _completions(self, full_text, offset, _timeout)
   1861         matched_text, matches, matches_origin, jedi_matches = self._complete(
-> 1862             full_text=full_text, cursor_line=cursor_line, cursor_pos=cursor_column)
   1863 

/usr/local/lib/python3.7/dist-packages/IPython/core/completer.py in _complete(self, cursor_line, cursor_pos, line_buffer, text, full_text)
   2029             completions = self._jedi_matches(
-> 2030                 cursor_pos, cursor_line, full_text)
   2031 

/usr/local/lib/python3.7/dist-packages/IPython/core/completer.py in _jedi_matches(self, cursor_column, cursor_line, text)
   1373         interpreter = jedi.Interpreter(
-> 1374             text[:offset], namespaces, column=cursor_column, line=cursor_line + 1)
   1375         try_jedi = True

/usr/local/lib/python3.7/dist-packages/jedi/api/__init__.py in __init__(self, code, namespaces, **kwds)
    725         super().__init__(code, environment=environment,
--> 726                          project=Project(Path.cwd()), **kwds)
    727         self.namespaces = namespaces

TypeError: __init__() got an unexpected keyword argument 'column'

During handling of the above exception, another exception occurred:

NameError                                 Traceback (most recent call last)
/usr/local/bin/odrivetool in <module>
    136             print("")
    137         import odrive.shell
--> 138         odrive.shell.launch_shell(args, logger, app_shutdown_token)
    139 
    140     elif args.command == 'dfu':

~/.local/lib/python3.7/site-packages/odrive/shell.py in launch_shell(args, logger, app_shutdown_token)
[... 90 lines cut ...]
-> 1684                 document, complete_event
   1685             ):
   1686                 complete_state.completions.append(completion)

/usr/local/lib/python3.7/dist-packages/prompt_toolkit/completion/base.py in get_completions_async(self, document, complete_event)
    268 
    269         async for completion in completer.get_completions_async(
--> 270             document, complete_event
    271         ):
    272             yield completion

/usr/local/lib/python3.7/dist-packages/prompt_toolkit/completion/base.py in get_completions_async(self, document, complete_event)
    194         Asynchronous generator of :class:`.Completion` objects.
    195         """
--> 196         for item in self.get_completions(document, complete_event):
    197             yield item

/usr/local/lib/python3.7/dist-packages/IPython/terminal/ptutils.py in get_completions(self, document, complete_event)
    114             except Exception as e:
    115                 try:
--> 116                     exc_type, exc_value, exc_tb = sys.exc_info()
    117                     traceback.print_exception(exc_type, exc_value, exc_tb)
    118                 except AttributeError:

NameError: name 'sys' is not defined

samuelsadok, 2021-01-18:
The stack trace looks as far as I can tell identical to https://github.com/ipython/ipython/issues/12745. It seems that the proposed workaround is this: https://github.com/ipython/ipython/issues/12745#issuecomment-751892538

mngr0, 2021-01-22:
Thank you very much,
I hope that this issue will be helpful for someone else in the future

closed 2021-01-22 by mngr0