odriverobotics/ODrive #549: Add enum values to json interface definition.

rowanG077, 2021-01-20:
I'm writing a Haskell type safe Fibre implementation. I really would like to have the enums available to me. Currently I use the `interface_generator.py` with my own template to generate the haskell enum data types. This is really not nice since everything else about the interface is in JSON that you can retrieve from the odrive,

I found a TODO that you guys would also like this enum to be transmitted with the odrive.

https://github.com/odriverobotics/ODrive/blob/ebd237673a483dec87b0d372dd09dfbf0bc7dd64/tools/enums_template.j2#L2

What is the format you guys propose? I want to try my hand at implementing this.

samuelsadok, 2021-01-21:
Regarding the Haskell implementation, have you seen [libfibre](https://github.com/odriverobotics/ODrive/blob/devel/Firmware/fibre-cpp/include/fibre/libfibre.h) that landed recently? I strongly recommend to bind against this library rather than making a pure Haskell implementation. This will make it easier to adapt to future changes in upstream Fibre (there are plans to upgrade the low level protocol in the not-so-far future).

Regarding your actual question, here's my take on it:
I would like to move away from the single monolithic JSON because it doesn't scale well. So I could imagine there being a separate JSON snippet for every enum type which would be similar to the format in "odrive-interface.yaml".

So something like an array of `{"name": ... "value": ... "doc": ...}` dictionaries, where `doc` is optional.

On a higher level, these JSON snippets would be retrieved with a remote function call that roughly has the signature
`array<enumerator_info> get_enum_info(fibre::Codec enum_codec)`.

Currently libfibre exposes codecs as strings in the `libfibre_subscribe_to_interface()`-callbacks, so that would need to be changed to expose them as `fibre::Codec` instead.

rowanG077, 2021-01-22:
I have looked at `libfibre` and I think it's too flexible to seriously work with. Any attribute/function can disappear/appear at any time... That's just not something that I can paper over with a sane abstraction.

Why was this approach chosen instead of opening a remote interface and then that interface describing what it can do? That way you can create a consistent abstraction over the remote object instead of always having to worry the remote object may change it's interface. 

I understand this is more flexible but this flexibility makes it very hard to work. And as far as I can see there is not much benefit for having this flexibility.

samuelsadok, 2021-01-22:
Yeah it's true that this level of flexibility might be a step too far. I also thought this when reconsidering the API.

An equally powerful alternative would be to make interfaces static and let the objects change interfaces altogether (i.e. from interface A to interface B). However dynamic interfaces is something that we don't use yet so it would be ok to add it back later.

So feel free to replace `libfibre_subscribe_to_interface()` with something like `libfibre_load_interface()` which returns a static list of attributes and functions.

rowanG077, 2021-01-29:
Which repo is the right now? Should I write a PR against this repo, or [this](https://github.com/samuelsadok/fibre) repo or [this](https://github.com/samuelsadok/fibre-cpp) repo?

samuelsadok, 2021-01-29:
https://github.com/samuelsadok/fibre (`devel` branch) is the main Fibre repo which also includes some CI tooling so you can create a PR on that or on https://github.com/odriverobotics/ODrive. The fibre-cpp repo is only intended as a mirror to make inclusion as subtree/submodule easier.

Wetmelon, 2021-09-13:
This has become a libfibre topic, closing.

closed 2021-09-13 by Wetmelon