odriverobotics/ODrive #557: nvm getting erased on connection issues

Jconn, 2021-03-29:
We have some odrives connected over usb. Occasionally on power cycle or when an odrive is sent a reboot command, the non-volatile memory will be erased.

We are currently running fw 0.4.11 

I have been looking into root causes for this recently and I see that maybe https://github.com/odriverobotics/ODrive/pull/441 is related. 

One thing I am curious about is the checksum validation. I see that the [checksum usage is optional](https://github.com/odriverobotics/ODrive/blob/fw-v0.4.11/Firmware/communication/ascii_protocol.cpp#L77) (I believe this is unchanged in devel). I am worried that the simplicity of the config erase command (it looks like just an "se" needs to be sent) coupled with an optional checksum is causing our issue. Does the `odrive` python module always send a checksum? I will experiment with making this checksum mandatory if so.

samuelsadok, 2021-03-30:
The checksum you're referring to is part of the human-readable ASCII protocol. However the `odrive` python module uses a binary machine-to-machine protocol (Fibre) to communicate with the ODrive.

Firmware v0.4.11 is quite old and some things got fixed since then (e.g. #371), so I would recommend checking if the issue still persists in the latest release.

Wetmelon, 2021-09-13:
Duplicate of #527

closed 2021-09-13 by Wetmelon