odriverobotics/ODrive #561: IPython warning

ketaro-m, 2021-04-10:
I'm using M1 mac and Big Sur 11.2.3, python version is 3.8.6 by pyenv and I installed odrive following [this instruction page](https://docs.odriverobotics.com/). 
When I start `odrivetool`, a warning as below shows up.

```
(.venv)$ odrivetool
ODrive control utility v0.5.1.post0
/opt/homebrew/lib/python3.9/site-packages/IPython/core/interactiveshell.py:937: UserWarning: Attempting to work in a virtualenv. If you encounter problems, please install IPython inside the virtualenv.
  warn("Attempting to work in a virtualenv. If you encounter problems, please "
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

In [1]:
```

I think this is because the IPython which is launched inside odrivetool is not the one installed in `pip install --upgrade odrive`, but the python3.9 in homebrew I installed before as the error message says. Just in case, I checked the IPython version as below and it is 3.9.4, which is not 3.8.6.

```
In [1]: import sys
In [2]: print(sys.version)
3.9.4 (default, Apr  4 2021, 17:42:23)
[Clang 12.0.0 (clang-1200.0.32.29)]
```

Since I thought it is something to do with paths python looks at, I reconstruct it.

```
$ source .venv/bin/activate
```
But another issue occur...

```
(.venv)$ odrivetool
ODrive control utility v0.5.1.post0
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/Users/ketarom/.pyenv/versions/3.8.6/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/Users/ketarom/.pyenv/versions/3.8.6/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/ketarom/sony/projects/odrive/.venv/lib/python3.8/site-packages/fibre/usbbulk_transport.py", line 191, in discover_channels
    devices = usb.core.find(find_all=True, custom_match=device_matcher)
  File "/Users/ketarom/sony/projects/odrive/.venv/lib/python3.8/site-packages/usb/core.py", line 1299, in find
    raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

In [1]:
```

This type of error is discussed [here](https://github.com/pyusb/pyusb/issues/120) as well. But I could solve is by putting homebrew's path top.

```
export PATH=/opt/homebrew/bin:$PATH
```

However, because homebrew path is prioritized now, homebrew IPython is launched as the first problem I mentioned.
How can I fix it?

Thank you.

samuelsadok, 2021-04-13:
It looks like `No backend available` in PyUSB is the main issue here and changing your `PATH` just has the unintended side effect of breaking the embedded IPython.

You could try to launch python in your virtualenv and manually run `import usb.core`, `usb.core.find(...)` to isolate the issue.

Otherwise you can try the newest (unreleased) version of odrivetool on devel. This no longer uses PyUSB but instead uses libfibre which directly links to libusb.

ketaro-m, 2021-04-13:
Thanks for your replay.
As you mention, it's probably the problem with PyUSB. Error as below occurred.

```
In [1]: import usb.core
In [2]: dev = usb.core.find()
---------------------------------------------------------------------------
NoBackendError                            Traceback (most recent call last)
~/odrive/.venv/lib/python3.8/site-packages/fibre/shell.py in <module>
----> 1 dev = usb.core.find()

~/odrive/.venv/lib/python3.8/site-packages/usb/core.py in find(find_all, backend, custom_match, **args)
   1297                 break
   1298         else:
-> 1299             raise NoBackendError('No backend available')
   1300
   1301     if find_all:

NoBackendError: No backend available
```

I changed python used in this project from pyenv python to `/opt/homebrew/opt/python@3.9`. Then the problem above was solved. I guess python under `/opt/homebrew/opt/` could somehow find `/opt/homebrew/opt/libusb` due to the path dependency. Is it right? 

```
(.venv)$ odrivetool
ODrive control utility v0.5.1.post0
Website: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

In [1]: import usb.core
In [2]: dev = usb.core.find()
```

I use python3.9 instead of python3.8 because homebrew-installed python3.8 fails in pip-installing numpy although the documentation says like below.
> As of Oct 2020, Matplotlib (required by odrivetool) had not been updated to work with 3.9, so please use 3.8.6.

Is that fine? I think Matplotlib have already gets to work with 3.9.

samuelsadok, 2021-04-13:
Yeah probably PyUSB doesn't find libusb when run from inside a virtualenv. I suggest to open an issue on https://github.com/pyusb/pyusb for this since this is out of scope of ODrive.

Odrivetool should work fine with Python 3.9. Matplotlib and numpy are not essential for odrivetool to work, so you can install odrivetool with `pip install odrivetool PyUSB --no-deps` to skip the unessential dependencies.

ketaro-m, 2021-04-14:
I see. Thank you for your kind reply!

closed 2021-04-14 by ketaro-m