odriverobotics/ODrive #568: IndexError when trying to do a dfu update on Windows 10

cinderblock, 2021-05-15:
**Describe the bug**
IndexError when trying to do a dfu update on Windows 10

**To Reproduce**
1. Buy new Odrive 3.5
2. Connect to power
3. Install Python 3.8.6
4. Run `pip install odrive`
5. Run `odrivetool dfu`
6. Install `libusb-win32` driver for "STM32 BOOTLOADER" with Zadig

**Expected behavior**
Update Odrive to latest firmware

**Desktop (please complete the following information):**
 - OS: Windows 10 x64 in Dveloper Mode
 - odrivetool Version (`odrivetool --version`): `v0.5.1.post0`

**Additional context**

I thought #509 was fixed but apparently not.

```log
C:\Users\camer>odrivetool dfu
ODrive control utility v0.5.1.post0
Waiting for ODrive...
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
~\AppData\Local\Programs\Python\Python38\Scripts\odrivetool in <module>
    141         print_version()
    142         import odrive.dfu
--> 143         odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
    144
    145     elif args.command == 'liveplotter':

c:\users\camer\appdata\local\programs\python\python38\lib\site-packages\odrive\dfu.py in launch_dfu(args, logger, cancellation_token)
    463     firmware = FirmwareFromFile(args.file) if args.file else None
    464
--> 465     update_device(device, firmware, logger, cancellation_token)
    466
    467

c:\users\camer\appdata\local\programs\python\python38\lib\site-packages\odrive\dfu.py in update_device(device, firmware, logger, cancellation_token)
    277         if otp_data[0] == 0:
    278             otp_data = otp_data[16:]
--> 279         if otp_data[0] == 0xfe:
    280             hw_version = (otp_data[3], otp_data[4], otp_data[5])
    281         else:

IndexError: array index out of range
```

I've also tried:

 - running odrive from git repo directly
 - Using DFU mode switch

PAJohnson, 2021-05-22:
This bug is not the same as the bug from issue #509. You mentioned buying a new ODrive 3.5 - ODrive 3.6 has been in production for 2 years now - is this a typo? This section of the code checks the hardware version of the device, so the hardware version is important.

Can you add:
```print(otp_data)```
above line 277 in `dfu.py` to see what the one-time programmable data is?

cinderblock, 2021-06-15:
I am using a v3.5 that hadn't been out of the box. Wasn't sure if the difference to v3.6 was relevant so I thought I'd include it.

This also happened on another Windows machine but I didn't take the time to check as you suggested.

I ended up using a Linux machine to flash the ODrive to v0.5.2. But now I've got some ... new version pains still.

Regardless, looks like the DFU has been revamped so maybe this issue is moot anyway. Closing

closed 2021-06-15 by cinderblock