odriverobotics/ODrive #572: Firmware for v3.5_48v ODrive does not connect to odrivetool/softbricks board

Fallstop, 2021-05-23:
**Background context**
After soft bricking my Odrive by trying to [re-flash it using Python 3.9](https://discourse.odriverobotics.com/t/odrive-not-being-recognized-by-windows/6720/3), the DFU tool did not recognise the version. So I went the more manual route and downloaded the firmware listed in the [releases channel](https://github.com/odriverobotics/ODrive/releases) to flash with the Odrive DFU tool to flash directly.

**Describe the bug**
The firmware for the Odrive 3.5 48v from the releases channel (Stable and RC's included) does not connect to the Odrive Tool but (I think this is the case) compiling `CONFIG_UART_PROTOCOL=native` instead of default `ascii` works as expected (I think that is the factor that made it work). 

**To Reproduce**
[Follow this guide here](https://docs.odriverobotics.com/odrivetool#device-firmware-update) under the heading `Upgrading firmware with a different DFU tool`
AKA:
Download the firmware of the releases tab on Github, and flash to an Odrive using any tool while the device is in DFU mode. Restart the device into RUN to get the `Device firmware update successful.` prompt and try to connect using the ODrive tool to see it magically not connect, and not respond to `odrivetool dfu`.

**Expected behaviour**
For it to connect after following the guide on how to flash the Odrive when soft-bricked.

**Desktop (please complete the following information):**
 - OS: Kubuntu 21.04
 - odrivetool Version: v0.5.1.post0
BTW I compiled it on GitHub workspaces.

**Potential Solutions**
a) Default the CONFIG_UART_PROTOCOL it so it can connect to the ODrive tool on the automatic releases
b) If it is intended functionality, please add a note to the [documentation for this](https://docs.odriverobotics.com/odrivetool#device-firmware-update) under the heading `Upgrading firmware with a different DFU tool`.

**Working firmware**
[Here is the firmware that I compiled for ODrive v3.5 48v](https://next.jmw.nz/s/PdsoXdPn9CmRTMq)

samuelsadok, 2021-05-26:
Firmware version did you upgrade to?
Is it possible that you had a non-erased configuration on the device before trying DFU?

If yes it's possible that the new firmware didn't realize that it must disregard the incompatible configuration and so it started in a bad configuration that made it hang.

Maybe you can try to do another DFU with the config erased or a "null"-DFU (updating to the same version that's already on the device). If the bug occurs in one of these scenarios then the hypothesis above is not a suitable explanation.

Fallstop, 2021-05-27:
I'm not sure of the original firmware version, other than it was over a year old, but the starting soft brick occurred with the Odrive DFU accessing an inbuilt library that doesn't exist in Python 3.9 just after erasing the firmware.

> Maybe you can try to do another DFU with the config erased
Doesn't erasing the firmware lose the configuration so that doesn't make sense?

But I will play around with re-soft bricking the Odrive, and come up with some more parameters.

Fallstop, 2021-06-02:
So with the configuration reset and ODrive working, the automatic DFU runs without any crashes on Python 3.8 but then crashes after a slow wait for the device repair:
```

Waiting for the device to reappear...
Traceback (most recent call last):
  File "/home/fallstop/anaconda3/envs/odrive/bin/odrivetool", line 143, in <module>
    odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
  File "/home/fallstop/anaconda3/envs/odrive/lib/python3.8/site-packages/odrive/dfu.py", line 465, in launch_dfu
    update_device(device, firmware, logger, cancellation_token)
  File "/home/fallstop/anaconda3/envs/odrive/lib/python3.8/site-packages/odrive/dfu.py", line 429, in update_device
    odrive.configuration.restore_config(device, None, logger)
  File "/home/fallstop/anaconda3/envs/odrive/lib/python3.8/site-packages/odrive/configuration.py", line 70, in restore_config
    with open(filename) as file:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/odrive-config-unknownserialnumber.json'
```

Even though the flash finished successfully, it still bricks the device, which I then need to bump into DFU mode and reflash with the firmware I compiled to get it working again.

Wetmelon, 2021-09-13:
Can you try flashing 0.5.3 firmware and using 0.5.3.post0 odrivetool?

closed 2021-11-20 by Wetmelon