odriverobotics/ODrive #573: Firmware v0.5.1 sends both native reponse and ASCII protocol error in response to native command over UART

Jascha-N, 2021-05-24:
**Describe the bug**
Some valid native protocol packets (or actually stream frames) trigger both the native protocol handler and ASCII protocol handler to send a response. This makes the native protocol over UART pretty much unusable, as the response can sometimes contain strings such as "unknown command\r\n" or "invalid command format\r\n", mixed in with the actual native response.

See this post for more information: https://discourse.odriverobotics.com/t/unknown-command-in-native-uart-mode/7093

**To Reproduce**
See the forum post for some insights on how to reproduce.

**Expected behavior**
The ASCII protocol handler should never respond to native protocol commands (starting with 0xAA). Alternatively, make it possible to disable the ASCII protocol.

**Desktop (please complete the following information):**
Not applicable. I have used a standard ESP32 board for communication.

samuelsadok, 2021-05-26:
The next firmware release will have a configuration option to set the UART protocol: https://github.com/odriverobotics/ODrive/blob/fw-v0.5.2rc1/Firmware/MotorControl/odrive_main.h#L73

That means the ASCII protocol and the Native protocol (aka Fibre) can no longer be enabled at the same time on the same UART, thus preventing these sort of issues.

Wetmelon, 2021-09-13:
0.5.3 has released with this feature.  Please reopen if it's still a problem.

closed 2021-09-13 by Wetmelon