odriverobotics/ODrive #574: Expose available parameters and ENUMS (as json) via ASCII protocol

pbrier, 2021-05-31:
**Is your feature request related to a problem? Please describe.**
To allow for backwards and forwards compatibility when using the ASCII protocol with the Odrive one should know the available parameters and enum values used in the firmware. Now this needs to be synced externally (via a known version of the drive fw and the associated interface file from that build). 

**Describe the solution you'd like**
Allow the client to interrogate the drive via the ASCII protocol to present the available parameters and enums (as a JSON file).
e.g. "sp\n" will respond with the currently embedded_json, augmented with the enums from "enums.py" 

**Describe alternatives you've considered**
Generating the JSON in the build process and distribute that in my own software (one file per version of the firmware, so 0.5.2.json, 0.5.1.json etc.) but this is error prone and not forward compatible (install a new odrive with different enums in your machine, and it may not function). This is specifically true for displaying the correct error flags.

**Additional context**
I would like to write a simple universal driver for the Odrive using the ASCII serial protocol, with minimal dependencies.

samuelsadok, 2021-05-31:
Is there a particular reason that you would like to use the ASCII protocol? The ASCII protocol is only intended for simple (quick & dirty) use cases. It is not expected and does not aspire to work well in more general machine-to-machine communication. For this there is the native protocol (aka Fibre) which does provide the feature you're requesting (apart from the enum transfer).

pbrier, 2021-06-01:
I would like to reduce the dependencies on the client side to a minimum. The libfibre dependencies are too much, and I would not like to re-implement the protocol from scratch. I just need to send velocity setpoints and retrieve the axis state and errors. I think the ASCII protocol suits a cyclic communication at ~ 100Hz (what I need).

samuelsadok, 2021-06-01:
> I would like to reduce the dependencies on the client side to a minimum.

I strongly agree with this, therefore the only runtime dependency of libfibre (apart from the OS) is libusb and on Windows libpthread because I haven't had time yet to remove it from the build process.

Prebuilt binaries are available as build artifacts on the [devel branch of fibre](https://github.com/samuelsadok/fibre/tree/devel), for instance [here](https://github.com/samuelsadok/fibre/actions/runs/755558483) for the latest commit.

If you're up for compiling the binaries yourself you could remove those dependencies too by statically linking to them (libfibre for Windows already includes a statically linked libusb).

It's likely that the JSON in it's current form will not remain in the firmware in the long term, therefore exposing it on the ASCII protocol would slow down development of more advanced protocol features that are not covered by the current JSON format.

pbrier, 2021-06-01:
Samuel, even a dependency on an USB stack or an OS is huge imo. Consider use cases where the host has no OS (an embedded controller or PLC for example) and only a serial connection. If there is no "lean and mean" serial protocol available, CAN would be the only other option (apart from reverting to GPIO/STEP/DIR/PWM). I think there are other users that value the ASCII protocol. But I could be wrong. In general I like devices to be "self describing" where possible. So an embedded description of endpoints and enumerated values would be great. Or standardized in such a way that it also works conveniently. E.g. CAN Open / Ethercat EDS files (but that is a huge effort).

samuelsadok, 2021-06-02:
Those dependencies I mentioned actually only apply to the precompiled binaries. You can embed fibre-cpp into embedded firmware without any OS (although the client side currently still uses dynamic memory allocation). It runs on ODrive after all. You can also use any a UART driver as a backend instead of USB (although this is not well tested).

From a developer and user perspective ideally there should be only one in-house designed protocol and that protocol is called Fibre and covers all use cases you describe. The reason we keep the ASCII protocol around is mainly for compatibility with very constrained devices (like Arduinos) and to reduce friction for rapid prototyping (cause Fibre doesn't have many language bindings yet). Think of the ASCII protocol like an Arduino: It's great for rapid prototyping and for that it's loved by many but expect limitations and don't use it in a product that you intend to sell.

When it comes to preexisting protocols we're open to anything. CANOpen is something we probably want to do at some point.

pbrier, 2021-07-01:
I understand the wist to support a single protocol and interface as a developer. However: developing motion control software AND a comprehensive protocol library for all different platforms and languages that is well supported and maintained is a daunting task! From my perspective: I appriciate an interface that is either (1) trivial to implement yourself in any language (e.g. basic ASCII GET/SET/EXEC commands, JSON responses) with minimal dependencies or (2) based on a well supported standard that has multiple implementations on different transports, platforms and languages. Based on another issue post here I looked in to UAVCAN and think this would suite the typical use-cases for integration in a Robotics, AGV or industrial system! micro-ROS, CanOpen, Ethercat and other options are great, but come with a much larger implementation effort. Just for you to consider! And once again: I value a simple ASCII protocol! They tend to last the longest  e.g. GCODE, NMEA, MODBUS ;-). Just my 2 cents...

closed 2021-07-01 by pbrier