odriverobotics/ODrive #575: Flashing fw-v0.5.2 on 3.6-24V board doesn't appear to work.

ryanpennings, 2021-06-03:
**Describe the bug**
I've tried to flash fw-v0.5.2 to my 3.6-24v boards which shipped with fw-v0.5.1-dev.
I've tried the odrivetool dfu, STM32CubeProgrammer and STLink with the same results.
I've also tried on Windows, MacOSX and Ubuntu 18.04.

When doing this, odrivetool reports firmware version 0.5.1 after a reporting device firmware update successful.
I was able to build the firmware from source and flash that - which reports the correct firmware version.

**To Reproduce**

With odrivetool dfu:

```
odrivetool dfu
ODrive control utility v0.5.2.post0
Waiting for ODrive...
Found ODrive 208F35853056 (v3.6-24V) with firmware v0.5.1-dev
Checking online for newest firmware... found v0.5.2
Downloading firmware v0.5.2...
Saving configuration to /tmp/odrive-config-208F35853056.json...
Configuration saved.
Putting device 208F35853056 into DFU mode...
Erasing... done            
Flashing... done            
Verifying... done            
Waiting for the device to reappear...
Restoring configuration from /tmp/odrive-config-208F35853056.json...
Could not restore axis0.controller.config.anticogging.calib_anticogging: 'anonymous_interface_139984393894416' object has no attribute 'exchange'
Could not restore axis0.controller.config.anticogging.cogging_ratio: 'anonymous_interface_139984393890368' object has no attribute 'exchange'
Could not restore axis0.controller.config.anticogging.index: 'anonymous_interface_139984393896576' object has no attribute 'exchange'
Could not restore axis1.controller.config.anticogging.calib_anticogging: 'anonymous_interface_139984393894416' object has no attribute 'exchange'
Could not restore axis1.controller.config.anticogging.cogging_ratio: 'anonymous_interface_139984393890368' object has no attribute 'exchange'
Could not restore axis1.controller.config.anticogging.index: 'anonymous_interface_139984393896576' object has no attribute 'exchange'
Some of the configuration could not be restored.
Configuration restored.
Device firmware update successful.
```

```
Connected to ODrive 208F35853056 as odrv0
In [1]: odrv0.fw_version_revision
Out[1]: 1
```

Built from source:

```
odrivetool dfu ODriveFirmware.hex
ODrive control utility v0.5.2.post0
Waiting for ODrive...
Found ODrive 205435773056 (v3.6-24V) with firmware v0.5.2-dev

Warning: you are about to flash firmware [unknown version] which is older than the firmware on the device (v0.5.2-dev).
Do you want to flash this firmware anyway? [y/N] y
Putting device 205435773056 into DFU mode...
Erasing... done            
Flashing... done            
Verifying... done            
Waiting for the device to reappear...
Device firmware update successful.
```
```
odrivetool
ODrive control utility v0.5.2.post0
^[[AWebsite: https://odriverobotics.com/
Docs: https://docs.odriverobotics.com/
Forums: https://discourse.odriverobotics.com/
Discord: https://discord.gg/k3ZZ3mS
Github: https://github.com/madcowswe/ODrive/

Please connect your ODrive.
You can also type help() or quit().

Connected to ODrive 205435773056 as odrv0
In [1]: odrv0.fw_version_revision
Out[1]: 2
```

**Expected behavior**
Firmware should update to v0.5.2

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - odrivetool Version (`ODrive control utility v0.5.2.post0`)

**Additional context**
n/a

samuelsadok, 2021-06-03:
This is a known issue, the v0.5.2 binary was compiled with the wrong version number. The `master` branch includes a fix for this and we'll probably rerelease the binaries some time soon.

Wetmelon, 2021-07-14:
Duplicate

closed 2021-07-14 by Wetmelon