odriverobotics/ODrive #578: firmware 5.2 softbricks single ODESC3.6

Kezii, 2021-06-06:
Firmware 5.2 from github releases renders ODESC3.6 unusable, to the point it doesn't show up as an usb device, while 5.1 worls

```
> sudo st-flash --reset write ODriveFirmware_v3.6-24V_5.1.bin 0x08000000
[...]
> lsusb | grep -i odrive
Bus 005 Device 035: ID 1209:0d32 Generic ODrive Robotics ODrive v3
```

```
> sudo st-flash --reset write ODriveFirmware_v3.6-24V_5.2.bin 0x08000000 
[...]
> lsusb | grep -i odrive
```

this is reproducible every time

madcowswe, 2021-06-06:
Hi,
We do not support unofficial hardware. Please contact the vendor of your board to see if they can fix the underlying issue for you.

closed 2021-06-06 by madcowswe

marja0900, 2025-08-21:
Is this true? Odrive bricks odesc? I thought it was open-source

marja0900, 2025-08-22:
Can you give me a straight answer. I don't mind paying I just need to know if it's open source of if I'm wasting my time.

samuelsadok, 2025-08-22:
Firmware for ODrive 3.6 is open source (this repository). Firmware for newer ODrive models (Pro/S1/Micro) is not.