odriverobotics/ODrive #581: Position deviation error

pheowl, 2021-06-08:
I am building a CNC/portal robot based on a home-built BLDC linear motor controlled by ODrive. Currently I interface the ODrive with a GRBL controller using step-dir mode to implement motion control from g-code (works). Since the GRBL controller has no way of knowing the position, it would be great if one of the GPIOs could be configured as axis-error. In particular, the pin should change state (go high/low) if the position error goes above a certain threshold. In that case, the motion controller can pause the g-code execution, raise an error and possibly save the operator or the machine/motor/work piece. 

I guess an alternative might be to use the CAN bus with LinuxCNC or something similar. That, however, will lead to a quite sizable increase in complexity in my application.

I guess it should be straight forward to implement. What do you think?

Thanks a lot.

Wetmelon, 2021-06-10:
You could do this, but it's unlikely that we'd implement it, it's just too niche.

madcowswe, 2021-06-18:
ODrive v4 will have a dedicated ERROR pin, and "tracking error" is an error type we want to implement in the future.
So with those combined, I think you could get this functionality on ODrive v4.

pheowl, 2021-06-23:
Thank you for the information. I will see how my project progresses (currently I am making only slow progress) and then decide whether I will try to implement it myself or wait for ODrive v4.

closed 2021-07-18 by Wetmelon