odriverobotics/ODrive #582: startup_closed_loop causes UNKNOWN_PHASE_ERROR

Wetmelon, 2021-06-10:
**Describe the bug**

Using startup_closed_loop to jump into closed loop mode causes an UNKNOWN_PHASE_ERROR but setting it manually does not.
https://discourse.odriverobotics.com/t/firmware-0-5-2-release-thread/7151/33?u=wetmelon

**To Reproduce**
Steps and configuration necessary to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10]
 - odrivetool Version (`odrivetool --version`)

**Additional context**
Add any other context about the problem here.

samuelsadok, 2021-06-11:
What are the steps to reproduce this bug?

I tried this and it works as expected (fw 0.5.2, ODrive v3.6):
```
odrv0.erase_configuration()
odrv0.axis0.config.startup_motor_calibration = True
odrv0.axis0.config.startup_encoder_offset_calibration = True
odrv0.axis0.config.startup_closed_loop_control = True
odrv0.save_configuration()
[ODrive reboots]
[ODrive calibrates]
odrv0.axis0.current_state # outputs "8" (AXIS_STATE_CLOSED_LOOP_CONTROL)
```

Wetmelon, 2021-07-14:
CNR

closed 2021-07-14 by Wetmelon