odriverobotics/ODrive #585: Firmware upgrade on Windows 10 with v0.5.2 fails after erasing

cinderblock, 2021-06-15:
### Description
Firmware upgrade on Windows 10 with v0.5.2 fails after erasing

### To Reproduce
Run `odrivetool dfu` on Windows 10 x64 with `v0.5.2` `odrivetool` on a `v3.6` (24V) board that is on `v0.5.1` firmware

### Expected behavior
DFU update successfully

### Desktop
 - OS: Windows 10 x64
 - odrivetool Version `ODrive control utility v0.5.2.post0`

### Additional context

Device was left without any flash. Would not enter DFU mode on its own.

Had to set DFU switch, reboot, and plug into Linux machine to flash with `odrivetool dfu`.

### Log

```log
...
Checking online for newest firmware... found v0.5.2
Downloading firmware v0.5.2...
Saving configuration to C:\Users\CHTACK~1\AppData\Local\Temp\odrive-config-2079368B424D.json...
Configuration saved.
Putting device 2079368B424D into DFU mode...
Erasing... done
Flashing... (sector 0/7)
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
~\AppData\Local\Programs\Python\Python38\Scripts\odrivetool in <module>
    146         print_version()
    147         import odrive.dfu
--> 148         odrive.dfu.launch_dfu(args, logger, app_shutdown_token)
    149
    150     elif args.command == 'liveplotter':

c:\users\chtacklind\appdata\local\programs\python\python38\lib\site-packages\odrive\dfu.py in launch_dfu(args, logger, cancellation_token)
    469     firmware = FirmwareFromFile(args.file) if args.file else None
    470
--> 471     update_device(device, firmware, logger, cancellation_token)
    472
    473

c:\users\chtacklind\appdata\local\programs\python\python38\lib\site-packages\odrive\dfu.py in update_device(device, firmware, logger, cancellation_token)
    395         for i, (sector, data) in enumerate(touched_sectors):
    396             print("Flashing... (sector {}/{})  \r".format(i, len(touched_sectors)), end='', flush=True)
--> 397             dfudev.write_sector(sector, data)
    398         print('Flashing... done            \r', end='', flush=True)
    399     finally:

c:\users\chtacklind\appdata\local\programs\python\python38\lib\site-packages\odrive\dfuse\DfuDevice.py in write_sector(self, sector, data)
    200             status = self.wait_while_state(DfuState.DFU_DOWNLOAD_BUSY)
    201             if status[1] != DfuState.DFU_DOWNLOAD_IDLE:
--> 202                 raise RuntimeError("An error occured. Device Status: {!r}".format(status))
    203
    204     def read_sector(self, sector):
RuntimeError: An error occured. Device Status: (0, 2, 0, 0)
```

samuelsadok, 2021-09-01:
@cinderblock Did this issue occur on every try on your Windows machine?
@jlhamilt is your device status the same (`Device Status: (0, 2, 0, 0)`)?

If this happens again (consistently), try to replace line 201 in file `DfuDevice.py` (`if status[1] != DfuState.DFU_DOWNLOAD_IDLE:`) with `if (status[1] != DfuState.DFU_DOWNLOAD_IDLE) and (status[1] != DfuState.DFU_IDLE):` and let me know how it goes.

You can find the full file path in the backtrace. In this case it's `c:\users\chtacklind\appdata\local\programs\python\python38\lib\site-packages\odrive\dfuse\DfuDevice.py`.

cinderblock, 2021-09-02:
@samuelsadok Yes. However I ended up using a linux machine to push forward instead.

samuelsadok, 2021-09-03:
Ok, would your setup still allow to retest on Windows?
If so, you can first check if the error still occurs on the Windows machine and if it does, add the fix that I suggested and try again. If you can confirm that the fix is effective then we can include it in the next odrivetool release.

cinderblock, 2021-09-03:
@samuelsadok I don't think so, at least not easily. All of my ODrives have been updated to latest firmware and most are in situ.

samuelsadok, 2021-09-06:
Ok I'll leave this open for a bit then. If anyone runs across this and finds the [proposed fix](https://github.com/odriverobotics/ODrive/issues/585#issuecomment-910056559) effective, please drop a comment.

Wetmelon, 2021-09-13:
We've decided to ask users to not use `odrivetool dfu` on Windows.  Please use the STMCube Programmer utility (see https://www.st.com/en/development-tools/stm32cubeprog.html and https://docs.odriverobotics.com/odrivetool#device-firmware-update)

closed 2021-09-13 by Wetmelon