odriverobotics/ODrive #586: Cannot cleanly exit with `odrive.find_any()` in v0.5.2

cinderblock, 2021-06-15:
### Description
When using the ODrive library's (`import odrive`) `find_any`, you cannot exit (on Windows 10).
Neither `sys.exit()` nor exceptions force an exit.

Worse still, `CTRL+C` doesn't work. You must kill the process externally.

This used to work just fine in `v0.5.1`.

### Steps to Reproduce
```python
import sys
import odrive

print("ODrive Version:", odrive.__version__)
print("ODrive Voltage:", odrive.find_any().vbus_voltage)
sys.exit() # Optional
ASDF # Or any (syntax) error
```

### Expected behavior
Printing version (`0.5.2.post0`) and voltage, then exiting.

### Actual behavior
After ~1 sec, prints:
```log
PS ...\folder> python.exe .\test.py
ODrive Version: 0.5.2.post0
ODrive Voltage: 12.02614688873291
Traceback (most recent call last):
  File ".\test.py", line 7, in <module>
    ASDF
NameError: name 'ASDF' is not defined
```
Doesn't exit. Cannot `CTRL+C`. Must kill with `pskill python`.

### Desktop
 - OS: Windows 10 x64
 - odrivetool Version: `0.5.2.post0`

cinderblock, 2021-06-15:
Note: `odrv0.reboot()` will let it cleanly disconnect and close, as far as the computer is concerned.

But that breaks what I'm actively doing with the ODrive... so, no thanks.

cinderblock, 2021-06-16:
I have also looked for an API to let someone control this and not found anything. But I'm new to Python so I likely missed something.

samuelsadok, 2021-06-21:
Thanks for reporting this. This is actually an issue for our firmware testing workflow too. I'll make sure to look into it.

cinderblock, 2021-06-21:
I think it has to do with how `find_any()` is implemented. Specifically the threading.

Looking around, I think I've found the fix [here](http://skybert.net/python/python-pytest-hangs-forever/). Submitting a PR.

samuelsadok, 2021-07-02:
Fixed as per https://github.com/odriverobotics/ODrive/pull/587#issuecomment-872946402

closed 2021-07-02 by samuelsadok